portapack-mayhem/mayhem-firmware #131: DeBruijn Tool - A simple OOK remote calculator

BladeRunner-51, 2020-08-13:
Hi,

I believe this was mentioned on Furrtek's Havoc github page :- 

https://github.com/furrtek/portapack-havoc/issues/111

According to furrtek, the code is present but no interface in place hence not added within the firmware.

A simple OOK remote brute force calculator would be nice/useful to feature to have.

Thanks for keeping this project alive, amazing work!

BladeRunner-51, 2020-08-25:
Thanks Dirk275 for highlighting the code. Guys any chance of this being added to the interface?

euquiq, 2020-09-18:
Hi guys! As you maybe know, I've been working on adding the existing De Bruijn algorithm (it was already there, from Havoc's firmware). The code is apparently OK, but the existing algorithm seems to be flawed. I am currently working on a new one. There is an ongoing test-drive / discussion on Mayhem's DISCORD chat server.

Maybe any of you guys that can test this ongoing work, will be available in there.

euquiq, 2020-09-20:
In short: First we need to be sure that the base OOK App is working fine, so we can round up the De Bruijn on top of it. At this juncture, there are doubts about the original OOK app, even if it is working at all. 

 If I may redirect your attention into our DISCORD server chat, there is a channel dedicated to the OOK app and De Bruijn. There are lots of tests to be done. 

Another contributor, Flx user, proposed some enhancements (in a PR in here) which are over my head, technically, and definitely needs some peer reviewing.

In my tests, when I compare the signal that some keyfobs I got down here, and the original Mayhem firmware, against his PR, the actual visual (and audio) representation of the signal from his PR is way different that the one generated by the keyfobs and the original OOK app. 

Since my knowledge is very little on this topic, I am at a loss: It may indicate that either his PR is breaking something, or actually the contrary, his PR is fixing finally the OOK App at its base, and now it is doing a better work (even better than the keyfobs).

But as said, this needs to be tested by users with actual OOK devices.

BreakSecurity, 2021-02-01:
Hi, any progress?

eried, 2021-02-01:
Probably not, or somebody would have claimed the bounty :)

zxkmm, 2022-02-19:
> Hi, any progress?

There's a half usable firmware made by [anzejarni](https://github.com/anzejarni)(If I pointed a wrong user pls correct me), avaliable in [Mayhem's Discord server](https://discord.gg/tuwVMv3).

lujji, 2022-12-20:
Hi, I believe this can be closed now
https://github.com/eried/portapack-mayhem/pull/754

gullradriel, 2022-12-21:
Then you should claim the bounty, but I don't know how it's working, maybe @eried can help / decide ?

eried, 2022-12-21:
I'm not 100% sure, but basically if @BladeRunner-51 which I guess started the bounty and other contributors don't comment, we can close it and @cusspvz would be able to claim the bounty in bountysource website

girs1982, 2022-12-31:
no work , i flash myporta h2 , black screen only

joyel24, 2023-05-26:
Is this issue solved ?

eried, 2023-12-05:
Hi! anyone who added money to this bounty, please try to open a dispute on paypal, check -> https://github.com/bountysource/core/issues/1586

eried, 2023-12-05:
> Then you should claim the bounty, but I don't know how it's working, maybe @eried can help / decide ?

:/ well... it was never possible to claim it, due bountysource

closed 2023-12-05 by eried