portapack-mayhem/mayhem-firmware #132: Incorrect saving

XSXpaolo, 2020-08-13:
if you save a frequency for example 402.700 to store it on the radiosonde section of the freqman dir, the data does not end up in the selected radiosonde.txt file, but overwrites the acars.txt file, losing the frequencies stored acars.

eried, 2020-08-13:
Does it happen all the time? Have you tried more combinations of categories?

XSXpaolo, 2020-08-13:
I tried a couple of times and found the frequency to store in acars.txt and the content of acars all lost
[image]
[image]
[image]

XSXpaolo, 2020-08-13:
I used the save key from inside the radiosonde application, I set the radiosonde dir and put the description, see now aaa because I tried again and confirmed ... it ends in acars maybe because the first freqman txt

euquiq, 2020-08-14:
I did try to replicate the problem, but couldn't make it happen. On my tries, the frequency got correctly saved in a SONDES.TXT file I created inside /FREQMAN for this purpose. 

ACARS.TXT was not erased. 

The frequencies I add on SONDES.TXT are correctly saved in there. 

I am not saying the problem does not occur, but I would ask for a detailed step by step description of it, to see where I am failing to reproduce the error.

XSXpaolo, 2020-08-16:
I have solved the problem ... everything works regularly and also well, evidently for who knows what strange saving it was going to overwrite on the first available file "acars.txt". I formatted the card returned to the original dir "Freqman" and now everything works .. sorry for the delay in looking for the problem, I have been busy testing other more important things related to Hackrf ... Problem and problem solved. You are Fantastic and thank you for your attentive attention on the problems of others to which you respond with sublime speed. Thanks again

eried, 2020-08-16:
Cool!

closed 2020-08-16 by eried