portapack-mayhem/mayhem-firmware #134: The answer from the vendor TOPBID trough WISH

reclaudio, 2020-08-13:
Erwin ,
          believe me or not , the vendor send me this image for solving the problem ......
Translation from Italian .

Dear Customer,
Thanks for your message and for the photos.
Please don't worry, I will try my best to help you solve it.
Please see the attached photo.
Check the link marked in red and update the software to make it work fine, okay please?
We apologize for the inconvenience.
If you have any questions or need help, please feel free to contact me, I will always do my best to help you.
Best regards
(attached is a photo with all the instructions)

Thank you for choosing Wish, for any doubts contact us again.

Wish Team
.............................................................................................................................................................................................................................................................
[image]

What would you think if you were in my position ...? ....? 
Want to switch to Discord ? 

Thank you .

Claudio Re

euquiq, 2020-08-13:
What's this ? @reclaudio what issue are you having with Mayhem firmware ? If you are referring to your earlier issue of the 2700mhz cap, then it was already answered and closed. Please do not spam this repository.

eried, 2020-08-13:
@reclaudio I have no idea what is that paper, it is nothing related with this firmware.

closed 2020-08-13 by eried