portapack-mayhem/mayhem-firmware #138: Problem upgrading the firmware

Messian59, 2020-08-15:
I updated my portapack with the Mayhem firmware and the screen went black.

eried, 2020-08-15:
Check this video carefully https://www.youtube.com/watch?v=_zx4ZvurgOs specially the DFU procedure

euquiq, 2020-08-15:
Any luck ?

Messian59, 2020-08-16:
No, before I uploaded the new firmware, my portapack showed an error message "Mode Guru M4/memory insufficient, that's reason I flashed my sdr. When I tried upload the Havoc firmware again, the sdr still in black. 
[image]
[image]
[image]
[image]

eried, 2020-08-16:
Record a video commenting the steps (en español si te acomoda más) you are following and how it fails. I think you haven't watched the video carefully.

Messian59, 2020-08-17:
here's video:

https://youtu.be/8tqIvIy9LZQ

eried, 2020-08-17:
I see, you are doing it ok. It looks like a hardware problem but only on the portapack. I am not sure what you can do to fix it, but ideally testing another portapack.

Messian59, 2020-08-17:
It could be in the future with portapack H2. How i know if my sdr is broken? because  when I use the SDRSharp, the audio noise sounds chopped.

eried, 2020-08-17:
Can you test with a known source? That part might be your computer being a little slow

eried, 2020-08-17:
About the H2, yes. It is a nice upgrade, however I would contact the vendor of the H1 to complain first. We have next to 8K downloads of the firmware and this is the first report of a problem like this, so it is clearly a bad unit

GullCode, 2020-11-03:
Having an H2 I did not saw anything strange like you (latest mayhem). The only downside is that if the battery is not fully charged when using is in hackrf mode I see some interference.

GullCode, 2020-11-03:
And I see you used your hackRF without an antenna on it... Which is clearly a way to damage components. Never do this !
Also: how long is your USB cable ? Is it armored ? The shorter the better, and armored helps a bunch (as per here: https://github.com/mossmann/hackrf/wiki/Tips-and-Tricks )

Messian59, 2020-11-23:
I think my sdr is broken, I will buy another one with h2

ArjanOnwezen, 2021-05-17:
@eried  Since @Messian59 indicates it's a hardware issue, this issue can beclosed.

closed 2021-05-18 by eried