portapack-mayhem/mayhem-firmware #142: ui_navigation: Better Footer

eried, 2020-08-15:
User @klockee added a nice footer with a clock, but it needs minor improvements:

- [ ] The first update of the seconds take too long (that is why in current version is masked as "XX")
- [ ] Tapping on the clock should pop clock configuration
- [ ] Inside "Settings", it should be possible to hide the clock

ArjanOnwezen, 2021-05-17:
#361 Added option to show/hide clock in UI settings

eried, 2021-05-18:
Did you explore the initial delay thing? it is quite hard to work without a debugger/emulator fixing those details :/

ArjanOnwezen, 2021-05-18:
No I didn't look at the initial delay. Also I haven't implemented a button for the clock. But maybe someone else finds this annoying enough to fix :)

ArjanOnwezen, 2021-05-19:
It would be good if you could have a look at the PR, so it can be merged. :)

eried, 2021-05-20:
Yes, I will check it in my portapack soon. I want to be sure the labels are not overlapping or something in the different places

ArjanOnwezen, 2021-05-31:
@eried any chance you have some time to check the PR?

closed 2023-12-04 by eried