portapack-mayhem/mayhem-firmware #146: units feet/meter cleanup

eried, 2020-08-17:
There are several instances on the firmware where there is an unit hard-coded as default. (i.e. on geomap)

We need to:

- [ ] move everything to metric?

or

- [ ] move everything to 1 unit specified in settings?

euquiq, 2020-08-17:
I think that there may be an explanation based on different "industry conventions, like "the aeronautic uses imperial" but strangely enough, the "radiosondes" use metric  ?

eried, 2020-08-19:
Yes, it might be the case on some, but nevertheless I am sure in some sections **feets** are being defaulted for another reason.

GullCode, 2020-11-08:
I vote for the metric system, because it is the only system of measurement with an official status in nearly every country in the world (https://en.wikipedia.org/wiki/International_System_of_Units )
And as you're talking about aeronautic, the NASA went using metric system since 2007 (https://www.space.com/3332-nasa-finally-metric.html)

eried, 2020-11-08:
"except where impractical." 🤣 that is a very open statement

GullCode, 2020-11-08:
Well, nearly only the US is using imperial metrics ;-)

jLynx, 2022-04-04:
We could add a global imperial/metric setting as like you say, there are a few apps that use units

eried, 2023-12-04:
I am closing suggestions without much discussion to keep issues clean. Feel free to re-open if there is a valid reason

closed 2023-12-04 by eried