portapack-mayhem/mayhem-firmware #151: can't build current head in gentoo

ZeroChaos-, 2020-08-19:
This is silly big, sorry in advance
```
zero@theprophet portapack-mayhem % sudo MAKEOPTS="-j1" ACCEPT_KEYWORDS="**" emerge portapack-mayhem -a1 --jobs=1                                             (git)-[master] 

These are the packages that would be merged, in reverse order:

Calculating dependencies... done!
[ebuild  N     ] net-wireless/portapack-mayhem-9999::pentoo  0 KiB

Total: 1 package (1 new), Size of downloads: 0 KiB

>>> Verifying ebuild manifests

>>> Emerging (1 of 1) net-wireless/portapack-mayhem-9999::pentoo
>>> Unpacking source...
 * Repository id: eried_portapack-mayhem.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_ERIED_PORTAPACK_MAYHEM
 *   EGIT_OVERRIDE_BRANCH_ERIED_PORTAPACK_MAYHEM
 *   EGIT_OVERRIDE_COMMIT_ERIED_PORTAPACK_MAYHEM
 *   EGIT_OVERRIDE_COMMIT_DATE_ERIED_PORTAPACK_MAYHEM
 * 
 * Fetching https://github.com/eried/portapack-mayhem.git ...
git fetch https://github.com/eried/portapack-mayhem.git +HEAD:refs/git-r3/HEAD
git symbolic-ref refs/git-r3/net-wireless/portapack-mayhem/0/__main__ refs/git-r3/HEAD
 * Repository id: mossmann_hackrf.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_MOSSMANN_HACKRF
 *   EGIT_OVERRIDE_BRANCH_MOSSMANN_HACKRF
 *   EGIT_OVERRIDE_COMMIT_MOSSMANN_HACKRF
 *   EGIT_OVERRIDE_COMMIT_DATE_MOSSMANN_HACKRF
 * 
 * Fetching https://github.com/mossmann/hackrf.git ...
git fetch https://github.com/mossmann/hackrf.git --prune +refs/heads/*:refs/heads/* +refs/tags/*:refs/tags/* +refs/notes/*:refs/notes/* +HEAD:refs/git-r3/HEAD
git update-ref --no-deref refs/git-r3/net-wireless/portapack-mayhem/0/hackrf/__main__ 43e6f99fe8543094d18ff3a6550ed2066c398862
 * Repository id: mossmann_libopencm3.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_MOSSMANN_LIBOPENCM3
 *   EGIT_OVERRIDE_BRANCH_MOSSMANN_LIBOPENCM3
 *   EGIT_OVERRIDE_COMMIT_MOSSMANN_LIBOPENCM3
 *   EGIT_OVERRIDE_COMMIT_DATE_MOSSMANN_LIBOPENCM3
 * 
 * Fetching https://github.com/mossmann/libopencm3.git ...
git fetch https://github.com/mossmann/libopencm3.git --prune +refs/heads/*:refs/heads/* +refs/tags/*:refs/tags/* +refs/notes/*:refs/notes/* +HEAD:refs/git-r3/HEAD
git update-ref --no-deref refs/git-r3/net-wireless/portapack-mayhem/0/hackrf/firmware_libopencm3/__main__ 6a301c81f4144d72fc881eb4fc7cc77aa1a32a67
 * Repository id: greatscottgadgets_gsg-kicad-lib.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_GREATSCOTTGADGETS_GSG_KICAD_LIB
 *   EGIT_OVERRIDE_BRANCH_GREATSCOTTGADGETS_GSG_KICAD_LIB
 *   EGIT_OVERRIDE_COMMIT_GREATSCOTTGADGETS_GSG_KICAD_LIB
 *   EGIT_OVERRIDE_COMMIT_DATE_GREATSCOTTGADGETS_GSG_KICAD_LIB
 * 
 * Fetching https://github.com/greatscottgadgets/gsg-kicad-lib.git ...
git fetch https://github.com/greatscottgadgets/gsg-kicad-lib.git --prune +refs/heads/*:refs/heads/* +refs/tags/*:refs/tags/* +refs/notes/*:refs/notes/* +HEAD:refs/git-r3/HEAD
git update-ref --no-deref refs/git-r3/net-wireless/portapack-mayhem/0/hackrf/hardware_gsg-kicad-lib/__main__ e02e7855c2f9b5653b958dab6d4e0d97eade7166
 * Checking out https://github.com/eried/portapack-mayhem.git to /var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update -->
   repository:               https://github.com/eried/portapack-mayhem.git
   at the commit:            c626d83c3bccc057859a1660196821d41914ecb1
 * Checking out https://github.com/mossmann/hackrf.git to /var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/hackrf ...
git checkout --quiet 43e6f99fe8543094d18ff3a6550ed2066c398862
GIT update -->
   repository:               https://github.com/mossmann/hackrf.git
   at the commit:            43e6f99fe8543094d18ff3a6550ed2066c398862
 * Checking out https://github.com/mossmann/libopencm3.git to /var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/hackrf/firmware/libopencm3 ...
git checkout --quiet 6a301c81f4144d72fc881eb4fc7cc77aa1a32a67
GIT update -->
   repository:               https://github.com/mossmann/libopencm3.git
   at the commit:            6a301c81f4144d72fc881eb4fc7cc77aa1a32a67
[... 880 lines cut ...]
 * 
 * Call stack:
 *     ebuild.sh, line  125:  Called src_compile
 *   environment, line 2536:  Called cmake_src_compile
 *   environment, line  892:  Called cmake_build
 *   environment, line  871:  Called eninja
 *   environment, line 1313:  Called die
 * The specific snippet of code:
 *       "$@" || die "${nonfatal_args[@]}" "${*} failed"
 * 
 * If you need support, post the output of `emerge --info '=net-wireless/portapack-mayhem-9999::pentoo'`,
 * the complete build log and the output of `emerge -pqv '=net-wireless/portapack-mayhem-9999::pentoo'`.
 * The complete build log is located at '/var/tmp/portage/net-wireless/portapack-mayhem-9999/temp/build.log.gz'.
 * The ebuild environment file is located at '/var/tmp/portage/net-wireless/portapack-mayhem-9999/temp/environment'.
 * Working directory: '/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999_build'
 * S: '/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999'
 * 
 * The following package has failed to build, install, or execute postinst:
 * 
 *  (net-wireless/portapack-mayhem-9999:0/0::pentoo, ebuild scheduled for merge), Log file:
 *   '/var/tmp/portage/net-wireless/portapack-mayhem-9999/temp/build.log.gz'
 * 
zero@theprophet portapack-mayhem %                                                                                                                           (git)-[master] 

```

ZeroChaos-, 2020-08-19:
I can't paste the whole log (github size restriction) so I randomly snipped the middle out.  If I happened to cut something important please let me know, and I'll post the full log somewhere for download.

ZeroChaos-, 2020-08-19:
if it helps, I'm building with gcc 9.3.1

ZeroChaos-, 2020-08-19:
Tests so far:

Build hackrf upstream firmware.  PASS
Build portapack upstream firmware. FAIL

Based on the file which is erroring, I think https://github.com/sharebrained/portapack-hackrf needs to sync with upstream for the fix.

jboone, 2020-08-19:
Fix on the sharebrained/portapack-hackrf repo, in a branch with a single commit:

https://github.com/sharebrained/portapack-hackrf/tree/gcc9_3_assert_redefinition

Unfortunately, I've spent so little time with git lately that whatever skills I had are gone. So I failed at formulating a pull request. I'll leave it to others to determine how to move this forward.

eried, 2020-08-19:
Cool, I imported that commit to https://github.com/eried/portapack-mayhem/tree/gcc9.3-assert-redefinition could you check that it solves your issue @ZeroChaos- ?

eried, 2020-08-19:
Uhmm.. this fails on docker :/

```
The image 252276_574377:latest already exists, renaming the old one with ID sha256:c5a2d508c954c5fd3261ac691b7ccc6b5cd729bf384cd46c00a41545413197e7 to empty string
Loaded image: 252276_574377:latest
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /opt/build/armbin/arm-none-eabi
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /opt/build/armbin/arm-none-eabi
-- The ASM compiler identification is GNU
-- Found assembler: /opt/build/armbin/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /havoc/build
Scanning dependencies of target baseband_shared
[  0%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj
[  0%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M4/vectors.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore_v7m.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/common/ARMCMx/nvic.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsys.c.obj
/havoc/firmware/chibios/os/kernel/src/chsys.c: In function 'chSysInit':
/havoc/firmware/chibios/os/kernel/src/chsys.c:122:21: warning: cast between incompatible function types from 'void (*)(void *)' to 'msg_t (*)(void *)' {aka 'long int (*)(void *)'} [-Wcast-function-type]
  122 |                     (tfunc_t)_idle_thread, NULL);
      |                     ^
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdebug.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chlists.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chvt.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chschd.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chthreads.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdynamic.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chregistry.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsem.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmtx.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chcond.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chevents.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmsg.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmboxes.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chqueues.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmemcore.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chheap.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmempools.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/test.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testthd.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testsem.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testmtx.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testmsg.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testmbox.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testevt.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testheap.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testpools.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testdyn.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testqueues.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testbmk.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/hal.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/adc.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/can.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/ext.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/gpt.c.obj
[... cut ...]
firmware/baseband/CMakeFiles/baseband_shared.dir/build.make:2102: recipe for target 'firmware/baseband/CMakeFiles/baseband_shared.dir/stream_output.cpp.obj' failed
CMakeFiles/Makefile2:1682: recipe for target 'firmware/baseband/CMakeFiles/baseband_shared.dir/all' failed
make[2]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
CMakeFiles/Makefile2:692: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
Makefile:313: recipe for target 'firmware' failed
make: *** [firmware] Error 2
Action failed: see logs above for details
```

jboone, 2020-08-19:
If I hadn’t failed to create a proper pull request, I would’ve mentioned that there are likely other uses of these two functions in code added by HAVOC. Sorry about that. You’ll need to search for calls to the two functions and rename those too.

eried, 2020-08-19:
Ohhh so dumb! I'll do it tonight

jboone, 2020-08-19:
LOL, that's not dumb! The code base is a twisted knot, and I knew at the very least the HAVOC record function would use the function I renamed. I planned to warn you, but I forgot. That's life. :-)

eried, 2020-08-19:
> LOL, that's not dumb! The code base is a twisted knot, and I knew at the very least the HAVOC record function would use the function I renamed. I planned to warn you, but I forgot. That's life. :-)

It works now, it was just in 1 another place :)

ZeroChaos-, 2020-08-19:
> > LOL, that's not dumb! The code base is a twisted knot, and I knew at the very least the HAVOC record function would use the function I renamed. I planned to warn you, but I forgot. That's life. :-)
> 
> It works now, it was just in 1 another place :)

Did you push, because git head still doesn't work for me...

ZeroChaos-, 2020-08-19:
/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/common/lpc43xx_cpp.hpp:96:13: error: expected unqualified-id before ')' token
   96 | inline void assert() {
      |             ^~~~~~
/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/common/lpc43xx_cpp.hpp:96:13: error: expected ')' before '?' token
   96 | inline void assert() {
      |             ^~~~~~

eried, 2020-08-19:
yes I did :O

[image]

eried, 2020-08-19:
Can you compile https://github.com/sharebrained/portapack-hackrf/tree/gcc9_3_assert_redefinition ?

ZeroChaos-, 2020-08-20:
Yes, https://github.com/sharebrained/portapack-hackrf/tree/gcc9_3_assert_redefinition compiles cleanly

Should I be building master for portapack-mayhem or some other branch?

ZeroChaos-, 2020-08-20:
Ahh, found it, https://github.com/eried/portapack-mayhem/tree/gcc9.3-assert-redefinition builds cleanly as well.

I beg of you to merge it into master and maybe tag a release for me? :-)

eried, 2020-08-20:
Nice and Yes, it will be included in 1.2.1, but that's scheduled for few weeks 🤪

closed 2020-08-20 by eried

ZeroChaos-, 2020-08-20:
I appreciate the fix branch, but can this please get pulled into master? I would love to help test, but right now I can't.