portapack-mayhem/mayhem-firmware #152: unable to build from tags

ZeroChaos-, 2020-08-20:
```
zero@theprophet portapack-mayhem % sudo ACCEPT_KEYWORDS="~amd64" emerge portapack-mayhem -a1 --jobs=1                                                        (git)-[master] 

These are the packages that would be merged, in reverse order:

Calculating dependencies... done!
[ebuild     UD ] net-wireless/portapack-mayhem-1.2::pentoo [9999::pentoo] 0 KiB

Total: 1 package (1 downgrade), Size of downloads: 0 KiB

Would you like to merge these packages? [Yes/No] yes

>>> Verifying ebuild manifests

>>> Emerging (1 of 1) net-wireless/portapack-mayhem-1.2::pentoo
 * portapack-mayhem-1.2.tar.gz BLAKE2B SHA512 size ;-) ...                                                                                                            [ ok ]
>>> Unpacking source...
>>> Unpacking portapack-mayhem-1.2.tar.gz to /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work
>>> Source unpacked in /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work
>>> Preparing source in /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2 ...
 * Working in BUILD_DIR: "/var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build"
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2 ...
 * Working in BUILD_DIR: "/var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build"
cmake -C /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build/gentoo_toolchain.cmake  /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2
loading initial cache file /var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build/gentoo_common_config.cmake
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /opt/gcc-arm-none-eabi/arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:48 (add_subdirectory):
  add_subdirectory given source "hackrf/firmware/hackrf_usb" which is not an
  existing directory.

-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- <<< Gentoo configuration >>>
Build type      Gentoo
Install path    /opt/gcc-arm-none-eabi/arm-none-eabi
Compiler flags:
C               -Os -pipe -frecord-gcc-switches
C++             -Os -pipe -frecord-gcc-switches
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Module          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Shared          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0

-- Configuring incomplete, errors occurred!
See also "/var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build/CMakeFiles/CMakeOutput.log".
See also "/var/tmp/portage/net-wireless/portapack-mayhem-1.2/work/portapack-mayhem-1.2_build/CMakeFiles/CMakeError.log".
```

Can you please include the full sources in the tags so they can be built?  Or am I supposed to be downloading more than just https://github.com/eried/portapack-mayhem/archive/1.2.tar.gz ?

eried, 2020-08-20:
Oh no! It's gentoo 😂

But yes, releases don't include modules https://stackoverflow.com/questions/34719785/how-to-add-submodule-files-to-a-github-release so I guess you need to do some extra work with .gitmodules file? I am not really sure. Another alternative is just git cloning (and init the modules) from the release tag.

ZeroChaos-, 2020-08-20:
given that github doesn't support this (nice find), it seems like the easiest solution is to ship the precompiled bins to my users.  I will do that, thanks for providing them.

closed 2020-08-20 by ZeroChaos-