portapack-mayhem/mayhem-firmware #153: WiFi SSID beacon

jamesshao8, 2020-08-20:
Has anyone watched this video?
https://www.youtube.com/watch?v=tCQaMrRuIqA

In 42:37. They used a simple microcontroller to transmit QPSK to simulate a wifi ssid beacon that can be receive by an iphone.

I think it will be cool if portapack can do that also.

at 44:18, he said binary shift keying at 11Mbit and get a 1Mbit DSSS?

Unfortunately, I didn't find opensource information on what they do.
I am testing with gr-ieee802-11 first to see if we can generate this beacon in gnuradio.

jboone, 2020-08-20:
HackRF / PortaPack could totally do that. I helped Dominic get the demo working the day before. :-) In that case, it involved using a GPIO to switch between two RF switches, one of which adds a 180-degree delay. However, you should be able to accomplish the same thing by transmitting a stream of quadrature samples where you alternate between samples 180 degrees out of phase, e.g. 1+0j and -1+0j, or 0+1j and 0-1j. I'd given thought to how to implement this: I'd be careful about choosing your sampling rate. To make it easy, use either 11 or 22Msps (which *should* be possible on the HackRF, but isn't normally operated that fast). And, of course, you need code to form valid packets and perform the chipping. However, what I vaguely recall about the chipping sequence, you may just be able to copy one of two 11-sample chunks, each representing a different symbol. Then you'd string together those chunks based on the baseband message you want to send.

I've sometimes wondered what Wi-Fi cards and operating systems would do if presented with 10,000 unique Wi-Fi beacons in one second...

jamesshao8, 2020-08-20:
Thanks for your suggestion. I am trying to generate ssid beacon through gnuradio to get more knowledge on wifi. Then will get back to the realization on portapack.

MAVProxyUser, 2022-04-06:
you guys ever make any progress on this? I'm interested in spoofing DJI DroneID packets that are detected by AeroScope units. They pick up drone location data encoded in Vendor IE tags from beacon frames.

jLynx, 2022-04-06:
If you want to be able to create fake ssids you can do it this way https://github.com/spacehuhn/esp8266_beaconSpam. But to answer your question, not implemented in the HackRF yet

MAVProxyUser, 2022-04-06:
Yeah ESP8266 is something I've already solved this on. I'm looking to make it more robust. 
Looks like scary + gr-802-11 can do it too. I'm not having much luck though.
https://www.bastibl.net/gnuradio-wlan-scapy/
https://github.com/bastibl/gr-ieee802-11/issues/119#issuecomment-623847844
https://archive.fosdem.org/2019/schedule/event/gr_scapy/attachments/slides/3366/export/events/attachments/gr_scapy/slides/3366/gnuradio_meets_scapy.pdf

eried, 2023-12-05:
Hi! @MAVProxyUser please try to open a dispute on paypal (I assume you added the 10 USD), check -> https://github.com/bountysource/core/issues/1586

closed 2023-12-05 by eried