portapack-mayhem/mayhem-firmware #154: More noise with H2 plugged in hackrf mode

Nuclrz, 2020-08-20:
Hey, I'm not sure if maybe this is just a problem with my portapack combo but I noticed when I'm in hackrf mode and using sdr# there is a lot of noise just scattered throughout the band. Unplugging the portapack fixes this however when using the built-in spectrum in mayhem the noise doesn't seem to be there. just want to know if this is a problem with the firmware or if anyone else has this problem.

reclaudio, 2020-08-20:
I have seen another problem .

When using the built-in spectrum in mayhem and I connect the USB port to the PC to avoid discharging batteries , the noise increase dramatically .

A: eried/portapack-mayhem
Cc: Subscribed
Oggetto: [eried/portapack-mayhem] More noise with H2 plugged in hackrf mode (#154)

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

XSXpaolo, 2020-08-20:
For the noise try to use a USB cable with ferrite rings: The ferrite rings are mainly used to attenuate the high frequency noise components, leaving practically unchanged the lower frequency ones which usually represent the useful signal to the system.
They are seen by the system as localized impedances with frequency dependent resistance and inductance and behave like the secondary of a transformer.

Nuclrz, 2020-08-20:
> For the noise try to use a USB cable with ferrite rings: The ferrite rings are mainly used to attenuate the high frequency noise components, leaving practically unchanged the lower frequency ones [...]

I think the noise is coming from the battery ic not the usb cable

eried, 2020-08-20:
Can you record a video with a example of the noise?

XSXpaolo, 2020-08-20:
Try to power the usb with 5volts from battery (6 vosts battery + 7805) or with a non-switcing power supply with low ripple and make sure it does not come from the usb cable with dirty power supply

euquiq, 2020-08-20:
I will introduce some noise into the issue :) maybe the whole charging / battery control circuit is introducing noise, at different levels, either under normal operation and specially -specially- while charging. It would be an "H2 problem". And maybe one of the motives why there is no actual charging circuit on the official H1 design ?

euquiq, 2020-08-24:
> I have seen another problem . When using the built-in spectrum in mayhem and I connect the USB port to the PC to avoid discharging batteries , the noise increase dramatically .

This is an H2 specific problem (which is non-official hardware, but rather a "kludge" from the chinese). Surely you'll get more noise ... there are extra ICs in there which were not part of the original design!

damadis, 2020-08-28:
I have noticed this as well with my H2. But my H2 came with a short "manual" that says at one point (original text):  "The usb just for charging the battery, or for the "hackrf mode". Can't usb as the power supply. There will be interference to the signal when charging, my no signal."

eried, 2020-08-28:
Can you record a video @damadis with this phenomenon? i.e. first hackrf alone, then hackrf with portapack, and third portapack with battery full (green led off/dim)

damadis, 2020-08-28:
Ok, video will be coming this weekend. In advance i already have something for you. I switched from H1 to H2 three months ago, and made some SDR# screenshots to compared the two devices. The first one shows my hackrf with portapack H1, and the second one is the same hackrf with the H2 on in. Antenna setup was the same. The three distinguish blocks in the waterfall in the second picture shows the change in the noisefloor as the  battery loading gets off --> on --> off.

[image]
[image]

eried, 2020-08-28:
Yeah I have seen that, but not on my device. On a H2 I saw that was even observable on the waterfall on the device itself.

What I do not know is why that charging IC continues with random blinks even when the battery is full (I would expect some periodic checking - 2 seconds- ). If you do a small demo we can get people testing the same to find clues (usually those IC control settings like delay and charge amp via resistors or caps).

reclaudio, 2020-08-31:
At this link you can find a video of what happens when you power the Portapack 2 with an external USB from a PC and from a power bank .The interference is radiated from the charging circuit and battery toward the stylus antenna : 
The video is not public , only  visible trough this link :
https://youtu.be/33n3nE3fEaQ

smannchen, 2020-08-31:
Perhaps the charging circuit is sharing a common ground with the sma, but that wouldn't improve when you disconnect the antenna. I think the antenna is just picking up rf energy from the charger. It should probably be shielded.

damadis, 2020-08-31:
I tested it on my PC and made a short video. HackRF without/with PP H2, battery charging on/off and in the end the H1.
[https://youtu.be/ijAmDUYjycE](url)

euquiq, 2020-09-01:
> I tested it on my PC and made a short video. HackRF without/with PP H2, battery charging on/off and in the end the H1.
> https://youtu.be/ijAmDUYjycE

Thanks for the video, Damadis. Very interesting and clarifying for me at least.

eried, 2020-09-01:
Nice video! thanks. I will check against my H2. Can the TCXO on the H2 also play part on this?

eried, 2020-09-01:
> I tested it on my PC and made a short video. HackRF without/with PP H2, battery charging on/off and in the end the H1.
> https://youtu.be/ijAmDUYjycE

I cannot replicate any of that. Mine does not change at all. I'll record and post a video probably tomorrow. I think you have a faulty unit.

eried, 2020-09-02:
Here is the video https://youtu.be/a_7Xc1_6-l4 I'm gonna close this issue, because it is some specific problem in your H2

closed 2020-09-02 by eried

pedroCX486, 2023-05-21:
https://github.com/eried/portapack-mayhem/issues/898#issuecomment-1556269511

Not unique to him. I have a similar issue.