portapack-mayhem/mayhem-firmware #158: APRS decoder

dqs105, 2020-08-21:
Maybe add an APRS decoder like qtmm?
There is already an AFSK receiver, and maybe adding a decoding layer on top of that will do.

jamesshao8, 2020-08-25:
Please add some more information about APRS. I will see what I can do.

texasyojimbo, 2020-09-16:
@jamesshao8 most APRS transmissions (on VHF and higher) use 1200 baud Bell 202 AFSK (Narrow FM) as the hardware layer,
In the United States/Canada/Mexico/"a few other random countries" APRS traffic is sent on 144.390 MHz but in Europe it's on 144.800 MHz and some other countries have different standards. So obviously we need the ability to adjust frequency although 144.390 or 144.800 would be good default settings. (I think APRS is more popular in the United States so I would urge 144.39).

AX.25 packets are always the data link layer.  The frames are unnumbered and it operates in unconnected mode.

There isn't really a true network/transport layer for APRS although there are some clauses at the beginning of each message that indicate the source as well as where to forward the message (for a minimally-acceptable implementation, you can probably ignore those -- we are building a receiver not a digipeater). Basically once you have the AX.25 frame decoded it should be human readable (more or less). 

Of course APRS has a list of fields and if you wanted to plot on a map you'd need to parse those.

We already have an AFSK receiver but it doesn't seem to do anything to decode frame headers and doesn't have a squelch feature. So we just see a bunch of random junk characters instead of a useful message.

There's plenty of documentation on the Internet about AX.25 -- it's an open-source standard that has been around for about 40 years. The main site for APRS is www.aprs.org.

Protocol documentation: http://www.aprs.org/doc/APRS101.PDF

There are also several existing implementations of APRS / 1200 baud AX.25 packet. Open-source ones include DireWolf, WinAPRS, APRSX, and the 1200AFSK tool in gqrx. Direwolf in particular has a lot of documentation on its internals and is open source.

texasyojimbo, 2020-09-16:
There is also 9600 bps packet / APRS but it's not used very much and can be ignored in 95 percent of use cases.

jamesshao8, 2020-09-16:
@texasyojimbo Thanks for your information. I will read documents and see how it would take to implement on portapack.

immodr, 2021-01-27:
APRS Decoder: it is a super, perfect idea.. I am waiting for.. !

eried, 2021-01-27:
> APRS Decoder: it is a super, perfect idea.. I am waiting for.. !

a big bounty would convince coders 😛

East2West, 2021-04-26:
@eried Can this be closed now?

eried, 2021-04-26:
Yes, did you tap on solve issue in bountysource? I am not sure if I should close it before that

East2West, 2021-04-26:
I think it's safe to now. I tried clicking solve before, but it's wanting this issue to be closed first.

closed 2021-04-26 by eried