portapack-mayhem/mayhem-firmware #159: Analog audio reciever frequency display not clean

dqs105, 2020-08-21:
When the frequency in "Analog audio" goes from 10Mhz to 9Mhz(also like going from 100Mhz to 10Mhz), there will be an extra garbage digit at the end of the numbers.

Steps to reproduce the behavior:
1. Go to Receivers -> Audio.
2. Tune frequency to 10Mhz.
3. Then tune to 9Mhz(or other frequency smaller than 10Mhz).

Expected behavior:
There should be only 4 digits after the decimal point.

Affected versions:
Version 1.2(Latest)

In older versions, the numbers are kept right. But in version 1.2, the numbers behaves differently, which resulted this bug.
[image]

eried, 2020-08-21:
Great finding :) thanks

dqs105, 2020-08-21:
I looked at the code, and found that firmware/application/string_format.cpp line 116-117 was modified, which resulted the formatting error. (Why modify this line? It works perfectly before.)

eried, 2020-08-21:
I am not sure, but feel free to send a fix 🎉

eried, 2020-08-21:
Ah I see based on your comment that it was modified for the scanner by @euquiq