portapack-mayhem/mayhem-firmware #164: Adding HAM Bands and Shortwave Bands

catalinalb, 2020-08-23:
Hello,

Please add the option to Listen to the HAM and Short Wave Radio bands. 
Would be ideal to integrate the FT8 digital modes as well.

Thank you,
Catalin

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Remember that adding stuff is always possible, but time is a limited resource for everyone. Check the wiki for more information how to compile the firmware and try to explore modifying the code yourself.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional**
If the suggestion is difficult to explain, additionally to the text please include images and videos.

Dirk275, 2020-08-24:
The HackRF is able to receive HAM bands from 160 m to 6 cm. What do you think is missing?

catalinalb, 2020-08-24:
The option to quickly select it in Mayham software / portapack

XSXpaolo, 2020-08-24:
There are plenty of digital systems for decoding
: cw, rtty, tor, gtor, packet, sstv in rx, fax, ft4, ft8, jt 4 9 and 65, pactor and many others .. there is a lot of digital stuff in amateur bands ... if it were up to me and if i knew how to use the cpp i would start with the more elaborate CW RTTY SSTV FAX PACKET and FT8 as decoding

catalinalb, 2020-08-24:
Yes I know, it doesn't seem to be a big deal, just I don't know how to program. Adding the HAM bands with a scanner function would be nice to have.

eried, 2020-08-24:
I do not think it is very realistic to ask for a functionality that it's just in your head. If you make some sketches and explain what exactly you really want, people might be able to help. 

In any case the quickest way to get this done would be to start from here:
https://github.com/eried/portapack-mayhem/wiki/How-to-collaborate#coding-new-stuff-or-fixing-bugs

Or contribute to your bounty here:
https://www.bountysource.com/issues/92376320-adding-ham-bands-and-shortwave-bands

XSXpaolo, 2020-08-24:
XSX(h1)Oggi alle 12:54
I share the previous post of Eried,but if the question is only to have the frequency of the amateur bands, you could add the following entries to the scanner.txt file inside the FREQMAN dir in order to have the band limits .... 
f=1800000,d=-
f=2000000,d=-
f=3500000,d=-
f=4000000,d=-
f=5330500,d=-
f=5405000,d=-
f=7000000,d=-
f=7300000,d=-
f=5330500,d=-
f=10100000,d=-
f=10150000,d=-
f=14000000,d=-
f=14350000,d=-
f=18068000,d=-
f=18168000,d=-
f=21000000,d=-
f=21450000,d=-
f=24890000,d=-
f=24990000,d=-
f=28000000,d=-
f=29700000,d=-

catalinalb, 2020-08-24:
Ideal would be to manually (or auto) scan only in the specific band. For example 40m band starting from 7000 kHz to 7300 kHz

https://en.wikipedia.org/wiki/40-meter_band

Also the Upper Side Band and Lower Side band should be autoselected according to each band. 
SDRUno software has this verry nicely organized. 

Thank you,
Catalin

closed 2020-09-14 by eried