portapack-mayhem/mayhem-firmware #165: Trunked radio rx support

xologram, 2020-08-24:
[Trunked radio](https://en.wikipedia.org/wiki/Trunked_radio_system) RX is used by emergency services (police, fire department etc) and since HackRF has nice wide bandwidth it can listen to trunked radio. The way it works with SDR#, for example, is with the help of [Unitrunker](http://unitrunker.com/) decoder, which decodes control channel which tells listening channel on which frequency to hop. If you want to try this with RTL-SDR you will need two dongles because they don't have bandwidth that's wide enough to cover both control and listening channels. HackRF can do that.

kingdagobert, 2022-04-02:
Hello Eried and collaborators ,
added 50$ more on this case and 15$ in the past on the same domain/category case >> https://github.com/eried/portapack-mayhem/issues/360 
really love to have voice decoding function related to trunked radio activities, will be very popular by the way.
Regards and full support

smannchen, 2022-04-02:
I believe this has been discussed and the portapack just doesn't have the processing power to decode digital voice.

bluegizmo83, 2022-08-09:
If it can't handle digital voice, how about Trunked analog voice? My area uses an EDACS96 analog voice Trunked system.

eried, 2023-12-04:
> Hello Eried and collaborators , added 50$ more on this case and 15$ in the past on the same domain/category case >> #360 really love to have voice decoding function related to trunked radio activities, will be very popular by the way. Regards and full support

@kingdagobert if you are still on time, please try to open a dispute for your money. Check: https://github.com/bountysource/core/issues/1586-