portapack-mayhem/mayhem-firmware #169: View the percentage of internal battery

ArcanA998, 2020-08-25:
Hi everyone, I own an H2 pack holder and therefore have an internal battery. 
The only problem is that I don't know when it is at 0% or near this threshold, in fact the only way to find out is to wait and see the black screen xD.
It would be useful to know the charge level with a percentage if it possible. 
(for example: 3.0v = 0% charge and 4.2v = 100% charge)
I would also like with a trivial voltmeter perhaps above the navigation bar.

Thank you all in advance for the replies !

smannchen, 2020-08-25:
I think this might be possible, but an easier option would be to put one of these on your case with a a little momentary switch. 
[image]

eried, 2020-08-25:
The battery is not directly connected, so this is not possible. But @smannchen is a good alternative. Another alternative would be to see if is possible to wire the battery power to some GPIO

ArcanA998, 2020-08-26:
Hi, thanks to both of the replies. I had already thought of putting an external mini display that marks the battery charge with its switch. The problem is that all this takes up internal and external space and I wanted to opt for a software solution.

I hope it will be possible in the future. Thanks again

eried, 2020-08-26:
There is probably smaller displays (remember those Duracell strips that indicated the remaining "power"?)

reclaudio, 2020-08-27:
You don’t have a continuous reading .

You have to push them to check the status of the battery .

A: eried/portapack-mayhem
Cc: Subscribed

 

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

XSXpaolo, 2020-08-27:
there is another alternative .. and even if it measures the voltage of 1.5v as a maximum, you can always add a resistor to calibrate the maximum charge reading for 4.2v https://www.thermographics.com/consumer/battery-tester

euquiq, 2020-08-27:
> Hi, thanks to both of the replies. I had already thought of putting an external mini display that marks the battery charge with its switch. The problem is that all this takes up internal and [...]

What about the reports that whenever the battery is being charged, it introduces noise / interference in the hackRF usage ? Since whenever you hook your usb cable to use your hackrf as "stand alone" the cheap charging circuit will kick in, then -if true- this would be a huge problem for h2 users.

Can anyone confirm this ?  (again: if true), in my (h1 fan) opinion: H2 users should rather dismiss the battery kludge altogether.

Furthermore (My h1 two cents): You should rather keep aligned to the "h2 spirit" and add an external, hardware based  solution. Touching the firmware to implement a battery ADC solution would be taxing to the system, for supporting a "cheap" kludge.

Having said that, if there is a GPIO with ADC capabilities free in there, -eventually adding two resistors so the measurable voltage complies with the adc capabilities- and a bit of code, should do it.

eried, 2020-09-14:
> Can anyone confirm this ? (again: if true), in my (h1 fan) opinion: H2 users should rather dismiss the battery kludge altogether.

Just as a reference for other people, there is no evident interference as seen here: https://www.youtube.com/watch?v=a_7Xc1_6-l4

I am closing this since it is not possible with the current hardware.

closed 2020-09-14 by eried