portapack-mayhem/mayhem-firmware #173: Bugs and Enchantments

ITAxReal, 2020-08-28:
Sorry for my bad English, but i think you can understand this text.

Bugs (I have H2 version):
Soundboard:
• Only first 100 files are listed
• When you try to go back from Soundboard mode with active playing, device going to dead freeze (only reset helping)
Other:
• Device generating noise when you tap on touchscreen (in TX mode)

Enhancements:
Soundboard:
• It will be awesome, if soundboard mode will show all files in WAV folder (if it's hardware limit's, then solution may be a paging function - load first 100 files, then if there are more files, navigate on it by switching page, or see next suggestion)
• Make a button of selecting folder with WAV files
• Sound Volume configuration
Scan:
• I can't find manual step configuration option
New functionality:
• Implement WAV player
• RX/TX audio mode - Combine Audio RX and Mic TX into one separated mode
• Be more stealth: FM receiver mode with tiny simplified interface, just frequency configuration between 60-120MHz and fake buttons like "Save radio station", "Search" (Say it's just a homemade FM radio receiver)
• More fool's stealth ideas: Simple game like Galaxian?? (It's my homemade pocket Dendy/Sega game)
• Battery level/voltage indication
• I can't find Frequency calibration function
• Receive car signalisation signals?

eried, 2020-08-28:
> • When you try to go back from Soundboard mode with active playing, device going to dead freeze (only reset helping)

Oh, nice finding, I think this should be easily fixable

> • Device generating noise when you tap on touchscreen (in TX mode)

Can you post a demo video so people can compare?

Dirk275, 2020-08-28:
> • I can't find Frequency calibration function

I cannot find it too. That was my issue #106 .

ITAxReal, 2020-08-28:
> > • When you try to go back from Soundboard mode with active playing, device going to dead freeze (only reset helping)
> Oh, nice finding, I think this should be easily fixable
Yeah, I think something like
`//Some code before closing
if(isPlaying){ stopPlay(); }
close();
//Some code after closing`
should be enough.
> > • Device generating noise when you tap on touchscreen (in TX mode)
> Can you post a demo video so people can compare?
Ok, just wait a few minutes

jsg700, 2020-08-28:
the calls options does not show digits only showing small red green yellow graph

ITAxReal, 2020-08-28:
> the calls options does not show digits only showing small red green yellow graph

I did not say anything about calls

euquiq, 2020-08-29:
My comments are my opinions only, of course. Here they are:

> Enhancements:
> • Sound Volume configuration

There is a sound volume configuration on every app that uses sound. Please clarify.

> Scan:
> • I can't find manual step configuration option *UPD: in Scan

Please clarify exactly what do you need ? There is already a selector for step configuration in the scanner.

> New functionality:

> • RX/TX audio mode - Combine Audio RX and Mic TX into one separated mode

it is already done ... the MIC TX RX app 

> • Battery level/voltage indication

It should be solved through hardware, not by software, since the battery addition is a "chinese thing" which is not validated by the official design... we should try not to do more "defacing" to the original portapack design

> • I can't find Frequency calibration function

Your Portapack already carries a very precise calibration clock, so the calibration function auto-hides, because there is no use for it.

ITAxReal, 2020-08-29:
> There is a sound volume configuration on every app that uses sound. Please clarify.

I mean output sound volume configuration (TX)

> Please clarify exactly what do you need ? There is already a selector for step configuration in the scanner.

Just selector, but what if I want to enter my own step?

> it is already done ... the MIC TX RX app

Sure, thanks for noticing

> It should be solved through hardware, not by software, since the battery addition is a "chinese thing" which is not validated by the official design... we should try not to do more "defacing" to the original portapack design

Yes, I know. The main question in this problem is "Can portapack detect input power voltage?"

> Your Portapack already carries a very precise calibration clock, so the calibration function auto-hides, because there is no use for it

Yeah, it's very similar to be truth

Dirk275, 2020-08-29:
> • I can't find Frequency calibration function
Your Portapack already carries a very precise calibration clock, so the calibration function auto-hides, because there is no use for it.

@euquiq :
It is clear that any device also with a good 0,5 PPM TXCO can of course be calibrated as I already said in issue #106 .
The standard function for the HackRF allowed a calibration on PPM basis, so it would not be helpful for a HackRF plus Portapack with it's 0,5 PPM TXCO.
What we would need now is a new calibration function on a 0,1 or 0,01 PPM basis, that replaces the old calibration function.

Good reasons for a calibration function for HackRF + Portapack:
1. Some RX functions need an exact frequency.
2. Most even cheaper SDR receivers like Airspy HF+,  ... allow calibration (though they also have precise TXCOs) and have internal storage for the calibration data.

ArjanOnwezen, 2021-12-07:
There are many things in here. Some ar le no real issues. Others are fixed. Like the soundboard freeze.
Closing ticket.
If there are additional wishes please create separate tickets for them with a good description.

closed 2021-12-07 by ArjanOnwezen