portapack-mayhem/mayhem-firmware #175: Support earphones with Volume Control

xyb, 2020-08-31:
I don't know if this is possible, but I think it would be great if the portapack firmware supported volume control via headphone cable control. There are very many such headphones. Apple and Android phones basically come with these headphones. I found specifications about them:
- https://source.android.com/devices/accessories/headset/plug-headset-spec
- https://electronics.stackexchange.com/a/242885

euquiq, 2020-09-10:
Thing is that the audio chip, the AK4951 inside the Portapack, AFAIK won't be able to measure the different resistive values that such headphones "Inject" into the MIC line for each pressed button. Simply put, the electronics inside the portapack are "not compatible" with that kind of headphones. Hopefully I am wrong ...

ArjanOnwezen, 2021-12-01:
closing ticket, since it seems to be a hardware limit

closed 2021-12-01 by ArjanOnwezen