portapack-mayhem/mayhem-firmware #177: Problem upgrading the firmware

sysardex, 2020-09-03:
I have a strange case and therefore there will be a strange question. Can I make portapack_h2 firmware in dfu mode? The problem is that my hackrf only works fine if I have firmware hackrf_one_usb.dfu. If I have firmware hackrf_one_usb.bin or portapack-h1_h2-mayhem.bin and all * .bin then it starts behaving strangely. It is constantly reset and disappears from the operating system.

eried, 2020-09-03:
How do you know the original one works normal? do you have a way to test if is actually transmitting or receiving?

sysardex, 2020-09-03:
> How do you know the original one works normal? do you have a way to test if is actually transmitting or receiving?

I have not checked the sending, but the reception on the original firmware is normal at all frequencies. I tested this through gqrx. On the portapack firmware, reception works only at 2150 - 2750 Mhz, also through portapack -> hackrf (usb) mode.
Also, when I tested the original firmware I did not remove the portapack board.

eried, 2020-09-03:
Strange phenomenon, I haven't seen it before. If you post a video with what works vs what does not it might help to find the "why" is different vs only hackrf

euquiq, 2020-09-04:
@sysardex, are you referring to the "original firmware" as to using the hackrf without the portapack ? You need to determine if it is only your portapack at fault, or both Hackrf AND portapack.

sysardex, 2020-09-04:
In the end, everything is very simple. I'll write, maybe someone can help. Portapack H2 behaves as I wrote before, if you connect it without a battery. The power supply of the usb port is not enough for it, also if you connect the 2A power supply.

eried, 2020-09-05:
Nice!

closed 2020-09-05 by eried