portapack-mayhem/mayhem-firmware #179: Spectrum Analyzer

jerrynsr, 2020-09-06:
A Spectrum Analyzer on Powerpak would be extremely useful.  One of the uses for me would be to chase down the interference we are experiencing at my r/c clubs airfield.  A HackRF One / Portapack is small enough to stuff in a club members  airplane. 
It would be great if the Spectrum Analyzer will record data to the SDCARD.

eried, 2020-09-14:
Could you explain a bit more exactly what interference do you want to track?

jerrynsr, 2020-09-14:
Interference with 2.4ghz r/c .   My r/c club is located in a remote rural area.  About 2 miles away from the T-Mobile erected a cell phone tower with what appears to be 8 backhaul dishes.  Several members noticed thier transmitters disconnect from thhier aircraft.  If they were lucky eventually they would reconnect.  Most of the time the reconnect would not happen in time and the aircraft would go in failsafe mode and crash in the woodlands.I suspect one of the backhaul dishes is making so much interference it is creating dead zones in our registered flight path because the recievers are getting saturated with noise.I plan on taking the HackRF One / Potapack up with my 3DR Solo on a GPS guided flight path that would be in the rf path of the backhaul dish that faces our field.  I plan on recording the output of my control transmitter and telemetry coming back from my aircraft and compare it to what is captured in the 2.4ghz region of the spectrum.

eried, 2020-09-15:
I see. I think it would be difficult to relate gps and the recorded values with some signal interference. Besides cellular uses around 1.9 ghz so maybe is not directly the main culprit? but I do not know.

Isn't better to:

1) Run your gps guided flight with one of the inputs connected to the normal receiver
2) Record that signal + gps
3) Set the transmitter in "range test mode" 

What I am saying is that even if you were able to record to the sd a long run of a broad spectrum sweep in the hackrf, it wont add much information to your problem.

jerrynsr, 2020-09-15:
How about just checking to see if there’s any rf  interference where my r/c equipment is operating on.    I don’t necessarily care about the gps location.
Whatever is broadcasting in the area is generating a lot of noise.  Not only r/c aircraft control at 2.4ghz but also GPS
I think there’s a bad piece of hardware on the tower that’s 2 miles away line of site of our field.

If it was just one club member with a problem I would say it’s the club member’s hardware.  But it’s affecting many of us and it appears that the disconnects
happen when our model aircraft are within line of site of the tower in question.   GPS lock also jumps around, not all the time but me and other member is the club 
have seen it too.

I just wanted physical evidence to prove it.  

From: Erwin Ried
Sent: Tuesday, September 15, 2020 4:15 AM
To: eried/portapack-mayhem
Cc: jerrynsr; Author
Subject: Re: [eried/portapack-mayhem] Spectrum Analyzer (#179)

I see. I think it would be difficult to relate gps and the recorded values with some signal interference. Besides cellular uses around 1.9 ghz so maybe is not directly the main culprit? but I do not know.
Isn't better to:
1. Run your gps guided flight with one of the inputs connected to the normal receiver
2. Record that signal + gps
3. Set the transmitter in "range test mode"
What I am saying is that even if you were able to record to the sd a long run of a broad spectrum sweep in the hackrf, it wont add much information to your problem.

gerryk, 2020-10-08:
This seems like a duplicate of https://github.com/eried/portapack-mayhem/issues/207

ShadowJonathan, 2022-07-09:
I'd want something more specific than #207, something that'd translate the wide spectrum view of information into something useful (i.e. which channels of wifi are being interfered the most)

gullradriel, 2023-05-10:
Done with latest app add/update.

closed 2023-05-10 by gullradriel