portapack-mayhem/mayhem-firmware #181: calls menu

kyumo, 2020-09-08:
1. Is it possible to add all gains controls to this menu?
2. it will be cool, if you can record all calls, and listen later

euquiq, 2020-09-10:
The CALLS application (not menu) has no audio. It is rather used for detecting close (like NEAR, in cm.) transmissions (i.e. from a keyfob). Regarding gains controls, there are already LNA and VGA controls exposed in there. The missing parameter is AMP (on / off).

 I just compiled a test version with AMP ON, and it did not have any apparent influence on detecting my keyfob. Since it is a "close call" concept perhaps this would explain NOT exposing the amp parameter... but it could be done: The app would need a bit of re-arranging, making space for the AMP 1/0 selector. 

I wonder if the calls app is already giving its best ... or it can be better. On my tests, detecting a keyfob has been "sketchy" (like not right away, and only 1 out 3 times giving a reasonably frequency).

What is your experience with it ? I will peek at the code today and try to add the AMP selector or at least have a better informed opinion about it.

kyumo, 2020-09-10:
For example, I tried with external antenna to monitor airband, and I found all channels in my location (of couse i knew it before), but It seems to me very useful. And I thought, that it willbe good also to listen all channels.

smannchen, 2020-09-10:
This is similar to the uniden close call feature of their scanners. Meant only for short distance. The scanner app already can do what you want, and probably more efficiently.

ArjanOnwezen, 2021-05-21:
@kyumo ticket will be closed.
Both @euquiq and @smannchen gave good motivations..

closed 2021-05-21 by ArjanOnwezen