portapack-mayhem/mayhem-firmware #186: Add Marine VHF radio in default sdcard txt

RobertoD91, 2020-09-13:
It would be nice to have Marine VHF frequencies in the default txt in FREQMAN folder

[https://en.wikipedia.org/wiki/Marine_VHF_radio#Channels_and_frequencies](url)

I don't know if it is better to have RX and TX in one file or separate, I am attaching the two versions of the files for convenience:

[VHF-RX.TXT](https://github.com/eried/portapack-mayhem/files/5214471/VHF-RX.TXT)
[VHF-TX.TXT](https://github.com/eried/portapack-mayhem/files/5214472/VHF-TX.TXT)
[VHF.TXT](https://github.com/eried/portapack-mayhem/files/5214477/VHF.TXT)

eried, 2020-09-13:
Hi, you can add them as a pull request :)

texasyojimbo, 2020-10-02:
I added some time and frequency stations (TIMEFREQ.TXT) since I didn't see them in any of the other lists.

As a ham radio operator, WWV has been keeping me entertained during times of bad propagation for over 20 years. :-p

closed 2020-10-02 by eried