portapack-mayhem/mayhem-firmware #188: Please add timer to Morse app so it can be used as a CW beacon

texasyojimbo, 2020-09-16:
**Is your feature request related to a problem? Please describe.**
I would like to use my HackRF portapack as a beacon while experimenting on 5.8 ghz and 10ghz. The current Morse app only seems to send my text string (my ham call) once. The Fox Hunt mode doesn't seem to do anything (it looks like there was originally a plan to add a timer to the Morse app but never got implemented by sharebrained).

**Describe the solution you'd like**
I would like to be able to set up the morse app to repeat a transmission on an adjustable schedule (every 10 seconds, every 1 minute, every 10 minutes, etc.). It might also be a good idea to add the option of a long tone after the text/callsign has finished sending (many ham beacons do that, for example, "vvv de n1abc n1abc <long tone>"

**Describe alternatives you've considered**
I could create an mp3 of my callsign and run it on repeat through soundboard, but this will send as NFM and not as CW. While I think this would be fine on the higher ham bands (uhf/shf) where band plans are not as strictly enforced, this would be a serious problem on HF where NFM is not allowed, and would also cause unacceptable levels of interference on 6 meters / 2 meters / 440 MHz where the weak signal bands are actively used.

strijar, 2020-09-22:
This is a good idea! I will do it

strijar, 2020-09-23:
Done - https://github.com/eried/portapack-mayhem/commit/4f2b3067842dd56135c54ec1305ac6ef47e971c5

eried, 2020-09-23:
Thanks @strijar

closed 2020-09-23 by eried