portapack-mayhem/mayhem-firmware #190: Record option for scanner

kyumo, 2020-09-19:
Can you add audio recording option in scanner? I would like to reacord all day activity.

GullCode, 2020-11-03:
Can you tell what you want to record ? The frequencies or the decoded signal on which the scanner stops ?
I mean, you want to have a file of a whole day containing frequencies, or just a wav file of what happened when the scanner caught it ?

kyumo, 2020-11-03:
I would like to have options, when I scan for example airband, to record all audio, when I listen one frequency - to record audio (And is it possible to have squelch in AM mode?)

pa0o, 2020-11-03:
thanks, pity that the missing antenna bias V   issue is not addressed yet

gullradriel, 2023-07-18:
Audio and Raw record have been added to Recon.
See the wiki for more information, or ask on discord.

closed 2023-07-18 by gullradriel