portapack-mayhem/mayhem-firmware #193: IMSI and SMS catching and detecting.

Firemission1343, 2020-09-21:
While looking into cemaxecuter work on DragonOS, and saw EFForg/crocodilehunter page, I thought that this idea may be feasible and was intrigued by the idea. Having a way to detect fake cell towers like a fox hunt would be neat. 

A way of catching IMSI and SMS numbers may be interesting for some as well...there are plenty of IMSI catching examples to look into. 

Any thoughts?

eried, 2020-09-22:
Did you got the imsi catcher working? can you share details?

Firemission1343, 2020-09-23:
I personally have only been able to do so on a Linux environment and not so on the hackRF firmware. To be honest, that is outside of my wheelhouse...I could put a bounty on it if anyone is interested?

eried, 2020-09-23:
A bounty will attract interested people :) but you need to provide some more details on the issue on what would mean it is "done". Maybe a quick sketch of what you expect in the UI, etc

closed 2023-12-04 by eried