portapack-mayhem/mayhem-firmware #194: Freezes when enabling Antenna Volage

miotislucifugis, 2020-09-21:
**Describe the bug**
Whenever I enable antenna voltage (from within an app, like Receive-> Audio)  the portapack freezes on the "enable DC Voltage" screen ( actually most of the screen is black- only the top of the "!" symbol is visible)     and needs to be reset.
Enabling DC Antenna voltage from the main menu works fine, and i can disable antenna voltage from with the apps...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Receive-> Audio'
2. Tap on 'Antenna voltage icon and then "yes" to enable Antenna DC voltage'
3. Screen freezes and needs to be reset.
** after some testing, Ive found that sometimes if I enable the antenna voltage before entering an app, then disable it from within the app, it will allow me to toggle it back on from with in the app w/o locking up.   Seems to only work once tho.  If i then disable it and try to reenable it a second time, it will freeze.

**Expected behavior**
I would expect to be able to toggle the DC antenna voltage on and off

**Affected versions**
using Latest "Mayhem" release v1.2.1

eried, 2020-09-22:
Hi, does it happen with older versions? https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

miotislucifugis, 2020-09-22:
It seems to work fine up thru and including mayhem_v1.2.
v1.2.1 is the only version that is giving me problems.

miotislucifugis, 2020-09-22:
I should note that, while the screen freezes and I cant get out with a reset,  the hackrf is still receiving, audio is still present and DC voltage does seem to be turned "on" because my external LNA gets activated.    the bug seems to be more  with the screen clearing/ "navigating back" process and not necessarily DC antenna voltage.
thanks btw.

eried, 2020-09-23:
Yeah, you are right. I think it might be related with some of the changes that @strijar made (i.e. https://github.com/eried/portapack-mayhem/commit/6f58c7929a96e68ba17d3045b0c9a3cef5019375) because if you launch for example the Receivers - TV first, and then Audio, it does not hang the screen. 

So it might be the "extended waterfall" or the speed. It needs further research.

strijar, 2020-09-23:
I have not been able to reproduce this issue on my device. Everything works as it should

eried, 2020-09-23:
> I have not been able to reproduce this issue on my device. Everything works as it should

:O interesting. If you turn it on, go to 'Receive-> Audio' and tap on 'Antenna voltage icon and then "yes" to enable Antenna DC voltage' and enable, everything works ok?

strijar, 2020-09-23:
Yes, everything works. The power turns on and off. I even checked all audio modes

euquiq, 2020-09-23:
This makes me think about that other user claiming that his H2 gives him electric shocks when touching the antenna. For a start, there seems to be several chinese h2 (and surely this would apply to chinese H1 too) faulty, and / or with subquality components (like the low sensitive touch screen -or in some cases even NO touchscreen at all installed).

Wouldn't it be possible that you may have one of those lemons,  @miotislucifugis  ?

miotislucifugis, 2020-09-23:
it is possible.  Mine is an H1 + hackrf combo from aliexpress.    Up until now, I havent noticed any issues, tho'.   Touch screen works reasonably well.     The DC voltage worked fine in previous releases.

 I did recently fry the internal RX LNA, but i think that was from static- (metal antenna touched my aluminum macbook and i heard a "pop"-  replacements are on the way)  Im not sure if these could be related.

euquiq, 2020-09-23:
@miotislucifugis sorry, I just downloaded Mayhem v1.2.1 on my H1, and had the same problem as you. I can confirm, also this bug. Somehow, one of the latest code additions seems to be the culprit. It surely will be found and fixed for the next release

miotislucifugis, 2020-09-23:
No worries-  Im just glad that it is code and not my hardware ;)

thanks for all the work you guys have done... happy that I could help in some small way.

closed 2020-09-24 by eried