portapack-mayhem/mayhem-firmware #197: Screen still freezing when enabling antenna voltage- mayhem 1.2.2

miotislucifugis, 2020-09-24:
**Describe the bug**
Unfortunately, Im still getting the screen hang when I enable antenna Dc voltage from within an app.    behavior seems consistent with before.   Will work if i enable the voltage before entering the app.... and then it lets me turn it off and then back on, once from within an app... but after that if will hang if i try to do it again.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Receive-> Audio'
2. Tap on 'Enable antenna voltage Icon- then tap "yes"'
3. Screen hangs

**Affected versions**
Using Mayhem 1.2.2 
Aliexpress H1 portapack & hackrf

miotislucifugis, 2020-09-24:
...Still works fine on Mayhem v1.2 and under ... so prob not my hardware, right?

eried, 2020-09-24:
Oh you are right, there is something else.

eried, 2020-09-25:
It should work fine now. I messed up a detail in the previous release :/

miotislucifugis, 2020-09-25:
yep. works. 
great work!

closed 2020-09-25 by eried