portapack-mayhem/mayhem-firmware #198: Voice tx as transceiver with AM modulation

AleyRobotics, 2020-09-25:
Voice tx as real transeiver with FM works great!
Is it possible voice tx as real transeiver with AM modulation?

JonaldJohnston, 2020-09-25:
AM or SSB?
they're kind of one in the same, but SSB is way harder to implement from what i understand

AleyRobotics, 2020-09-25:
> SSB

full AM modulation https://en.wikipedia.org/wiki/Single-sideband_modulation

strijar, 2020-09-28:
I'm interested. I will try to implement this

texasyojimbo, 2020-09-30:
This seems to be related to #84

strijar, 2020-10-07:
I started implementing LSB/USB transmit

AleyRobotics, 2020-11-11:
is there any progress or solution?

strijar, 2020-11-11:
Yes, i wrote an implementation of the SSB modulator. It is working (https://github.com/strijar/portapack-mayhem/tree/ssb-tx)

It remains to add the choice of the mode in GUI. But I don't like how "Miс TX" is implemented now. I want to rewrite it to be in the style of a Amateur transceiver (;

jumbo77, 2020-11-14:
Very Useful for HAM~  
Just Add HF Ampifire to work HF Transceiver

DomTheDorito, 2020-11-14:
> Yes, i wrote an implementation of the SSB modulator. It is working (https://github.com/strijar/portapack-mayhem/tree/ssb-tx)
> 
> It remains to add the choice of the mode in GUI. But I don't like how "Miс TX" is implemented now. I want to rewrite it to be in the style of a Amateur transceiver (;

Can we get a button to enable it at least while you rework the UI so we can test it out?

gerryk, 2020-11-15:
> Yes, i wrote an implementation of the SSB modulator. It is working (https://github.com/strijar/portapack-mayhem/tree/ssb-tx)
> 
> It remains to add the choice of the mode in GUI. But I don't like how "Miс TX" is implemented now. I want to rewrite it to be in the style of a Amateur transceiver (;

This looks good. At a quick glance it looks like it might be possible to replace the encoding in SSTV_TX with a selectable mode rather than the hardcoded FM that it currently in there. Currently the image processing and the encoding are folded in together but it could probably be done.

AleyRobotics, 2020-11-16:
Tested at lab. Works great! Default Gain is too high. Will test it at field conditions.

strijar, 2020-11-17:
Now I have little time to do this, but of course I will continue. There is an idea to make an alternative application "TRX" for testing modulators and UI. Leave the current "Mic TX" unchanged for now. (;

iz2fly, 2021-01-26:
Congratulations, I've been waiting for this for a long time.
Today I used HackRF_One with Portapack H2 in 1.2GHz, I connected an ham radio friend at 160Km in FM mode, excellent signal.
SSB is greater than 6 dB, that would be excellent!!!
[image]

gerryk, 2021-01-27:
I think now that a bounty has been offered, we should decide on some Acceptance Criteria for this solution, since the initial comment is a little light on specifics.
So, let me start...

**Acceptance Criteria**

- Mode selection exists in Mic Tx App UI to allow selection of the following modes 'AM', 'USB', 'LSB', in addition to that already available.
- Selection of a mode in the Mic Tx App UI causes transmissions to be modulated according to the mode selected.
- Existing functionality of Mix Tx App should remain unchanged, unless infeasible

iz2fly, 2021-01-27:
A little tip if possible, PTT should be as always, button pressed ON, button released OFF, but that's okay too.

aldude999, 2021-02-02:
> Yes, i wrote an implementation of the SSB modulator. It is working (https://github.com/strijar/portapack-mayhem/tree/ssb-tx)
> 
> It remains to add the choice of the mode in GUI. But I don't like how "Miс TX" is implemented now. I want to rewrite it to be in the style of a Amateur transceiver (;

Good work on getting the Hilbert transform in there. I got it up and running but I found one issue. LSB worked but USB modulated both sides of the carrier. It looks like to flip between USB and LSB, you should set your hilbert.execute(sample, q, i) for USB, or hilbert.execute(sample, i, q) for LSB, and leave the re = q and im = i for both. I assume flipping re and im around is causing the phase to be at the wrong angle to null the other side out.

iz2fly, 2021-02-02:
Great job strijar, gerryk, aldude999 and all the others.
I would like to try how it works, how to do it? It's possible?

strijar, 2021-02-02:
aldude999: I tested all three modes using the receive on the transceiver. Everything worked correctly. Perhaps you confused USB and DSB? The spectrum flipping is done on lines 61-64 in the dsp_modulate.cpp file

aldude999, 2021-02-02:
@iz2fly  I have it implemented in mic_tx along with the proper RX for each mode, and a selectable FM RX bandwidth (something I realized I wanted last time I was using it for an actual purpose) I should be able to upload it tonight.

@strijar I pulled the code into the proc_mictx.cpp and viewed it on an FFT output from another SDR and did not see LSB suppression when USB was selected. I also did not see a DSB option in those lines. It's possible I did something wrong when pulling it over, but flipping the i and q for the hilbert transform seemed to solve the issue at least.

strijar, 2021-02-02:
@aldude99: Hm... It is not enough to change only proc_mictx.cpp. There are a lot of changes.: https://github.com/eried/portapack-mayhem/compare/next...strijar:ssb-tx

See file firmware/baseband/dsp_modulate.cpp 

		switch (mode) {
				case Mode::LSB:		re = q;		im = i;			break;
				case Mode::USB:		re = i;		im = q;			break;
				default:			re = 0;		im = 0;			break;
			}

aldude999, 2021-02-03:
https://github.com/aldude999/portapack-mayhem I have a compiled binary release to test AM/SSB/DSB modulation in the microphone app here.

iz2fly, 2021-02-03:
> https://github.com/aldude999/portapack-mayhem I have a compiled binary release to test AM/SSB/DSB modulation in the microphone app here.
> 
> This has a long way to go in regards to improvements since I kind of just wanted to get everything working so people could check it out.

Thank you very much for this achievement, I have waited many years to see this application! Now I can do some field tests.
Thanks to all the developers, now only the details are missing.
I don't know if there are ham radios in the team, otherwise I can suggest a more suitable graphic.
Thanks so much again!!!

http://iz2fly.altervista.org/hackrf_one/

gerryk, 2021-02-03:
Hey aldude
I tried your build and it's working pretty well, even for a first drop!
Mic gain seems a tiny bit high, but that could be my crappy mic. Tx on both sidebands seems good... need to figure out the exact carrier offset, as I was just using an AOR8200 to test receive.

Other issues are mainly UI related... widget order is a bit all over the place, but that seems to be the case whenever any UI is edited.

I have just trred USB & LSB, but results very promising.

/ Gerry

> 
> 
> https://github.com/aldude999/portapack-mayhem I have a compiled binary release to test AM/SSB/DSB modulation in the microphone app here.
> 
> This has a long way to go in regards to improvements since I kind of just wanted to get everything working so people could check it out.

strijar, 2021-02-04:
@iz2fly Yes, I don't like the current design of this app either. I was planning to completely redo it. Perhaps I will return to this question soon. It is interesting to listen to your wishes. 73 de R1CBU (:

sliceratwork, 2021-02-04:
@iz2fly is that antenna bracket 3D printed or bought? Also, that's a log antenna right?

iz2fly, 2021-02-04:
@sliceratwork Yes, 3D file antenna bracket now in my website.

The signal in SSB mode compared to FM is very low, but we expect improvements, perhaps not for the software but for the hardware.

aldude999, 2021-02-11:
SSB and AM TX power are directly linked to mic gain, so that plays a big part. I think more options than the .5, 1.0, 1.5, and 2.0 might be useful. Or some kind of simple compressor could be written to keep the audio at a reasonable level.

iz2fly, 2021-02-12:
> SSB and AM TX power are directly linked to mic gain, so adjusting that will give you more or less power on the sidebands. I think more options than the .5, 1.0, 1.5, and 2.0 might be useful. Or some kind of simple compressor could be written to keep the audio at a reasonable level.

Thanks for the reply, however, already now there are excellent results with USB transmission. 160 km with 1.2 GHz and 110 km with 2.3 and 5.7 GHz.
I've been thinking about a new amateur radio interface, but maybe it's still early and I don't know if it's okay.
[image]

iz2fly, 2021-02-22:
I'm doing great tests in USB mode.
I connected a small amplifier, SPF5189Z. The amplifier is connected with coaxial transfer switch, I would like to control the coaxial transfer switch with "Turn ON bias voltage".
Is it possible to turn on bias voltage only in transmission mode and not during reception?

Radiomix2000, 2021-03-21:
@aldude999 
Hi, I tried to compile your repo, and it seems the code doesn't have all necessary files. I have a lot errors related with dsp_sos and some IIR filters. Can you explain, what's wrong?
Here is what I mean:
[console.txt](https://github.com/eried/portapack-mayhem/files/6177828/console.txt)

aldude999, 2021-03-21:
> @aldude999
> Hi, I tried to compile your repo, and it seems the code doesn't have all necessary files. I have a lot errors related with dsp_sos and some IIR filters. Can you explain, what's wrong?
> Here is what I mean:
> [console.txt](https://github.com/eried/portapack-mayhem/files/6177828/console.txt)
> 
> Also there is significant change in hackrf CPLD firmware (check Mossman repo). And now, to fix slow switching of TX/RX is necessary to comment out in firmware/application/radio.cpp the code
> 
> ```
>  //else {
>                 //      if( !hackrf::cpld::load_sram() ) {
>                 //              chSysHalt();
>                         //}
>                 //}
> ```

Looks like I left out `dsp_iir.cpp` and `dsp_iir.hpp` in firmware/common from [strijar's code](https://github.com/strijar/portapack-mayhem/tree/master/firmware/common). I'm going to try and rework my version to use his code more faithfully as it looks cleaner, and try and merge it with the new pushes before it gets too far behind.

aldude999, 2021-03-22:
New version has been pushed to [my repo](https://github.com/aldude999/portapack-mayhem) with the new changes from this repo added. Tested the code and it looks good so far, new release is there as well. I tried to clean up the ui_micproc code to follow the order of the items better, along with fixing the RX audio listening button that I accidentally shrunk before. Also changed the message format to only require 4 boolean values, AM, DSB, USB, and LSB instead of passing some of the unused variables I used for testing previously.

Radiomix2000, 2021-03-22:
> > @aldude999
> > Hi, I tried to compile your repo, and it seems the code doesn't have all necessary files. I have a lot errors related with dsp_sos and some IIR filters. Can you explain, what's wrong?
> > Here is what I mean:
> > [console.txt](https://github.com/eried/portapack-mayhem/files/6177828/console.txt)
> > 
> > Also there is significant change in hackrf CPLD firmware (check Mossman repo). And now, to fix slow switching of TX/RX is necessary to comment out in firmware/application/radio.cpp the code
> > 
> > ```
> >  //else {
> >                 //      if( !hackrf::cpld::load_sram() ) {
> >                 //              chSysHalt();
> >                         //}
> >                 //}
> > ```
> 
> Looks like I left out `dsp_iir.cpp` and `dsp_iir.hpp` in firmware/common from [strijar's code](https://github.com/strijar/portapack-mayhem/tree/ssb-tx/firmware/common). I'm going to try and rework my version to use his code more faithfully as it looks cleaner, and try and merge it with the new pushes before it gets too far behind.
> 
> Also I found my AM signal is not as strong as it could be, did some math and found out only modulating I or Q generates a signal at half strength. I figured modulating one would be the same as modulating both, but after testing it on a calculator I think I might be wrong as it resulted in a 50% modulation. Complex math is hard. :)

I will check Gnuradio repo. There is AM modulator in gr-analog. So at least it can be helpful to implement AM for portapack.
AM has 100% symmetrical spectrum, so is it right to just copy real stream to imaginary?

aldude999, 2021-03-22:
yes, an identical real and imaginary value should result in AM modulation.

ultraohm, 2021-04-03:
@aldude999 great job all AM SSB modes works great with HackRF and Portapack H2,
However will it be possible to implement VOX (auto) and RX audio Listening in the Microphone window when we are in AM SSB mode?
Thanks

iz2fly, 2021-05-06:
Too complex to use the Microphone menu with separate RX and TX frequencies and fixed frequency steps, when you're ready the caller has already changed frequency.

gerryk, 2021-06-02:
Since #84 is closed, this should also be closed, as they were effectively dups.

ArjanOnwezen, 2021-06-02:
> Since #84 is closed, this should also be closed, as they were effectively dups.

Hi @gerryk  This one still has a bounty on it, so not sure if this one can be closed.

gerryk, 2021-06-03:
> 
> 
> > Since #84 is closed, this should also be closed, as they were effectively dups.
> 
> Hi @gerryk This one still has a bounty on it, so not sure if this one can be closed.

Ok, cool. I haven't really looked into is so I'm not sure how the bounty mechanism actually works

iz2fly, 2021-06-27:
Test TX FM vs SSB
[image]

iz2fly, 2021-06-27:
Test TX FM
[image]
 vs SSB

iz2fly, 2021-06-27:
Test Tx FM vs SSB

[image]

texasyojimbo, 2021-06-28:
iz2fly -- how is the ssb signal being modulated?

is that a measure of just the suppressed carrier or the suppressed carrier
plus some modulation? SSB signal power is proportional to modulation.

iz2fly, 2021-06-28:
The measurement is very simple and practical.
It is a measure made as every ham radio does with his own radio.
With voice (oooooola) or whistle, usually the emitted power is the same as in the FM mode.

See photo of the signal in FM mode 2300MHz
[image]

aldude999, 2021-06-28:
FM will always output full power regardless of how loud the input signal is. SSB is highly dependant on the microphone gain, so by turning that up, you are also turning the signal output up as well. If you are not giving it a signal at full volume, you won't get an output at max power.

teixeluis, 2021-06-28:
@aldude999 I observed here:

https://github.com/eried/portapack-mayhem/issues/368

that in AM the carrier level is pretty low (I would expect that without modulation, FM and AM carriers should not differ much in power..)

In SSB even with modulation, the power output is comparatively low.

Cheers

iz2fly, 2021-06-28:
> FM will always output full power regardless of how loud the input signal is. SSB is highly dependant on the microphone gain, so by turning that up, you are also turning the signal output up as well. If you are not giving it a signal at full volume, you won't get an output at max power.

Thank you aldude999 for your comments. 
The vocal output was at full volume and with maximum gain and with various microphone capsules, maybe in SSB mode you need a different microphone than in FM mode? Or do you need an amplified microphone?
the result is that in FM mode the signal reaches the listener, in SSB mode this is much much lower. Usually this does not happen with any other transceiver. Unfortunately I am not an expert and I cannot answer. Mine is just a factual finding.
If this gain in SSB mode somehow can't be improved, that's okay, I'll put a signal amplifier on the output.

Brumi-2021, 2022-01-02:
Hi thanks to all of you for this great Mic TX App ! 
It has everything, combining  TX/RX, with full mod. Types FM/AM/DSB/SSB , and including Mic-VOX control !!!
Really amazing! Congratulations to all contributors !
I am also ham operator , and was looking for it ! 
Pls check my latest  PR ,  just small  fine tuning ,https://github.com/eried/portapack-mayhem/pull/457

iz2fly, 2022-09-30:
After a long time I picked up the HackRF-One.
Really amazing! Congratulations to all contributors! I have found many improvements in the Microphone APP.
- Improved audio.
- Step frequency to 10 Hz! Very good!
- Moving between the menus of the APP
- Other...
I still have to replace my U25 amplifier chip which has blown but I will do it soon.
If I can suggest:
- the best optimization would be to lock the Transmission frequency with the receiving frequency (TX = RX).
This change would greatly facilitate use as a QRP Radio.
Thanks from all the ham operators.

[image]

Brumi-2021, 2022-09-30:
Hi @iz2fly , thanks . 
I could see that you missed another important one , the empty space after MIC gain , will have Auto Level Mic Control (in AK4951 audio codec ) or Mic Boost control (in WM8731 audio codec) . Which one do you have ? Stay tuned , coming soon  I will realise new PR with both integrated options.  (Now pending latest checkings ) 
Cheers!

iz2fly, 2022-10-01:
Hello @Brumi-2021 
I really like these new updates. I have been absent for too long, even the battery needs to be changed. I am using the latest official firmware version 1.5.4. 
Are there any new updates already? Where can I find my audio codec?
Cheers!

Brumi-2021, 2022-10-01:
Hi @iz2fly 
If you ho to the "debug" menu , you will see which peripheral devices you can explore. In the audio block , you will see the audio codec IC mounted on your Portapack board ; AK4951 or WM8731. 

by the way . You can install this hot advanced test fw , related to issue #683

iz2fly, 2022-10-02:
Hi @Brumi-2021 
I did a short test with Nightly Release 220920 night, only listening and transmitting without tools, in RX mode no problem in FM and USB mode, the audio in USB mode is also very good.
In TX mode the signal is a bit distorted.
I am using the AK4951 audio codec.
I had another Portapack (now broken unfortunately )and it was better, maybe it had another audio codec. See audio test without modulation on my website.
With this Portapack in USB mode, pressing the PTT produces an annoying background noise.
Thanks.

Brumi-2021, 2022-10-02:
Hi @iz2fly 
No , that nightly release  that you tested , (is from 20th of sept) and I think it has not integrated the mic Automatic volume Level Control for the AK4952 Audio code that I released yesterday as a PR . 

You can download it from my last post of issue #683 or let me add also here the same link https://github.com/eried/portapack-mayhem/files/9690624/portapack-h1_h2-mayhem_WM_mic_boost_and_AK_ALC.zip

you will recognise in the menu that top ALC option (in AK4951)
[image]

Or  that BOOST option in the WM8731 
[image]

iz2fly, 2022-10-03:
Hi @Brumi-2021.
Perfect, I tried the 840da4c6 relase, very good modulation (ALC-OFF-12KHz).
Now for us ham radio it is enough to block the TX frequency with the RX frequency. We hope to have it soon.
Thank you very much.
[image]

Brumi-2021, 2022-10-03:
Hi @iz2fly ,
And did you try all the other ALC options ? 
One easy way to test it , Is to set up up AUTO vox PTT  control , reduce the vox control threshold to 15 (example) and then , when talking and you activate the vox control , and from that TX time , you can easily change the ALC options , and you will hear in your closer  receiver the real time effect .  (But in the high gain options you will need to use headphones to avoid sound coupling ) 
Let me know if you followed me 
Cheers , 73

GullCode, 2022-10-03:
> 
Hello. I'm not in favor of locking tx and Rx frequencies. A lot of repeaters are not using the same tx and rx freq, and I think it's better to have the possibility to choose instead of having a locked one.
Maybe adding two buttons to set 'tx from rx' and 'rw from tx' freq could do the trick.

Brumi-2021, 2022-10-03:
Hello @GullCode , thanks  for your post .
I fully agree with you . 
Maybe , as you suggested , a good compromise would be  a new button to communize TX/Rx Freq YES/NO 
but before continuing with it , I want to see the progress and merge of the latest PR . (ALC-ak/ BOOST-wm)
Step by step 
Cheers

iz2fly, 2022-10-03:
Hi @Brumi-2021. 840da4c6 relase. 
I have tried with two different microphones, the best setting is:
GAIN: 47 (Amp: 0, still broken) Mode: USB or FM
MIC-GAIN: x1 ALC -9dB-6KHz or
MIC-GAIN: x1.5 / x2 ALC -12dB-6KHz. or
MIC-GAIN: x1.5 / x2 ALC OFF-12KHz.
In TX mode the screen turns off.
I have tried various TX Activation settings, it does not work, it turns on but does not turn off by itself.
In AUTO mode, Audio RX listening does not work.
Maybe you need to test away from the radio, tests to do.
Cheers

GullCode, 2022-10-03:
@iz2fly Tx is turning the screen off because '[Stealth Mode](https://github.com/eried/portapack-mayhem/wiki/stealth-mode)' mode is enabled on your portapack. Check the icon on the top bar on the screen, there should be kind of a man with a hat in green, tap it to turn it grey.
TX should now keep the screen on.

[12 more replies]

closed 2023-03-21 by gullradriel