portapack-mayhem/mayhem-firmware #199: Cannot compile with GCC10

olewales, 2020-09-26:
Trying to compile firmware with gcc 10 yields following error:

```C
(...)
/home/olewales/portapack-mayhem/firmware/baseband/spectrum_collector.cpp: In instantiation of 'typename T::value_type spectrum_window_hamming_3(const T&, size_t) [with T = std::array<std::complex<float>, 256>; typename T::value_type = std::complex<float>; size_t = unsigned int]':
/home/olewales/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:142:79:   required from here
/home/olewales/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:114:28: error: non-constant condition for static assertion
  114 |  static_assert(power_of_two(s.size()), "Array size must be power of 2");
      |                ~~~~~~~~~~~~^~~~~~~~~~
/home/olewales/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:114:28: error: 's' is not a constant expression
/home/olewales/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:115:19: error: 's' is not a constant expression
  115 |  constexpr size_t mask = s.size() - 1;
      |                   ^~~~

(...)

```

Code may be made to compile by removing constexpr and assertion but there should be a better way to fix this

```diff
diff --git a/firmware/baseband/spectrum_collector.cpp b/firmware/baseband/spectrum_collector.cpp
index b43daadb..2f1a4084 100644
--- a/firmware/baseband/spectrum_collector.cpp
+++ b/firmware/baseband/spectrum_collector.cpp
@@ -111,8 +111,8 @@ static typename T::value_type spectrum_window_none(const T& s, const size_t i) {
 
 template<typename T>
 static typename T::value_type spectrum_window_hamming_3(const T& s, const size_t i) {
-       static_assert(power_of_two(s.size()), "Array size must be power of 2");
-       constexpr size_t mask = s.size() - 1;
+       //static_assert(power_of_two(s.size()), "Array size must be power of 2");
+       const size_t mask = s.size() - 1;
        // Three point Hamming window.
        return s[i] * 0.54f + (s[(i-1) & mask] + s[(i+1) & mask]) * -0.23f;
 };
```

I believe this is related to a bug fixed in GCC10: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=66477

GCC version:

```shell
$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/10.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-10.2.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (Arch Repository) 
```

eried, 2020-09-28:
I am not sure about this, I use docker so I actually do not deal with the compilation directly. If you find a solution, please add it to the wiki in a new section at the end of https://github.com/eried/portapack-mayhem/wiki/Compile-firmware

ZeroChaos-, 2021-04-30:
hi!  My portapack works with 1.3.1 but not with 1.4.0.  I figured no big deal, I'll simply bisect the issue, except I can't build because of this bug.  Pretty please fix build with gcc 10

```
[522/620] /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++ -DCORTEX_USE_FPU=TRUE -DTHUMB_NO_INTERWORKING -DTHUMB_PRESENT -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/baseband/. -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/common -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/os/ports/common/ARMCMx/CMSIS/include -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/os/ports/common/ARMCMx -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/os/ports/GCC/ARMCMx -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M4 -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/os/kernel/include -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/test -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/os/hal/include -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios-portapack/os/hal/platforms/LPC43xx_M4 -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios-portapack/os/hal/platforms/LPC43xx -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios-portapack/boards/PORTAPACK_BASEBAND -I/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/chibios/os/various  -mcpu=cortex-m4 -O3 -g -falign-functions=16 -fno-math-errno --specs=nano.specs -ffunction-sections -fdata-sections -fno-common -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -std=c++17 -fno-rtti -fno-exceptions -Weffc++ -Wuninitialized -Wall -Wextra -mno-thumb-interwork -mthumb -DTHUMB   -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHACKRF_ONE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -D_RANDOM_TCC=0 -D'VERSION_STRING="local-f161e85f"' -DLPC43XX -DLPC43XX_M4 -D__NEWLIB__ -DHAC [... cut ...]
      |                ~~~~~~~~~~~~^~~~~~~~~~
/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/baseband/spectrum_collector.cpp:116:28: error: 's' is not a constant expression
/var/tmp/portage/net-wireless/portapack-mayhem-9999/work/portapack-mayhem-9999/firmware/baseband/spectrum_collector.cpp:117:19: error: 's' is not a constant expression
  117 |  constexpr size_t mask = s.size() - 1;
      |                   ^~~~
ninja: build stopped: subcommand failed.
```

GullCode, 2021-04-30:
> I am not sure about this, I use docker so I actually do not deal with the compilation directly. If you find a solution, please add it to the wiki in a new section at the end of https://github.com/eried/portapack-mayhem/wiki/Compile-firmware

Done

ZeroChaos-, 2021-05-03:
Was building with gcc 10 fixed?  seems kind of short sighted to close this bug, it's not like gcc 9 will live forever, and gcc 11 is already released and being used by some distros....

eried, 2021-05-03:
oh, it can stay open in that case. Sorry about that.

Brumi-2021, 2021-10-02:
Any news about it ? I am using Docker under windows and it compiles well , but the process is really slow, it takes many minutes any new test  . It would be great to fix the compile errors with gcc 10 from Pentoo distro .

ArjanOnwezen, 2021-10-02:
Just compile using Linux or a VM on Windows  running Linux. Kali works fine.

Brumi-2021, 2021-10-11:
Hi,  thanks to all ,  who contributed to that good HackRf Portapack Mayhem  firmware . 
And thaks to  olewales  for your comment and hints about that  Compiler error , when changing GCC 9 -->  GCC 10.
I am newbie in C++ arm  embedded  firmware, but eager to learn .

I could reproduce exactly the same GCC 10 compiler error that Olewales pointed  ,(on  Pentoo (Gentoo distro) that from long time ago,  uses GCC 10.)

1-) The compiler error appears only  retated to that file :baseband/spectrum_collector.cpp,

2-) At compiler  time, in the code  we are defining 3  different  values of  spectrum Windowing in time  template type T :
               static typename T::value_type spectrum_window_none
               static typename T::value_type spectrum_window_hamming_3
               static typename T::value_type spectrum_window_blackman_3.
But  the  code ,  in line same file  , spectrum_collector.cpp, 145 ,  we are  only using and calling  that  spectrum window hamming 3, that is why also the declaration structure is  similar ,  the compiler , only  claims error about  the hamming_3 ,  lines 116, and 117 . 

3-) Now  in the current  Portapack Mayhem fw  , we have  two types compiler errors (when using GCC10) , 

 **a-) error: 's' is not a constant expression  ,
     117 |  constexpr size_t mask = s.size() - 1;**

After checking information about that "constexpr" 

"The primary difference between **const and constexpr variables** is that the initialization of a const variable can be deferred until run time. **A constexpr variable must be initialized at compile time.** ... A variable can be declared with constexpr , when it has a literal type and is initialized."

 then I believed that,  as you said (Olewales )   , it can be sort out ,  changing  constesxpr --> to const .
           **_const size_t mask = s.size() - 1;_**  

b-) Regarding  the  second error , 

**“In instantiation of 'typename T:**:value_type spectrum_window_hamming_3(const T&, size_t) [with T = std::array<std::complex<float>, 256>; typename T::value_type = std::complex<float>; size_t = unsigned int]':
/opt/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:145:79:   required from here
/opt/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:116:28: 

**error: non-constant condition for static assertion
  116 |  static_assert(power_of_two(s.size()), "Array size must be power of 2");”**

I  also tried to gather some  information about it ,  

--------------------------------------------------------------------------------------------------------------
"The compiler examines the static_assert declaration for syntax errors when the declaration is encountered. 
The compiler evaluates the constant-expression parameter immediately if it does not depend on a template parameter. Otherwise, the compiler evaluates the constant-expression parameter when the template is instantiated. 
Consequently, the compiler might issue a diagnostic message 
           * once when the declaration is encountered, 
           * and again when the template is instantiated.  (⇒ we have that case .)

And  checking  the static_assert declaration sintax : 
static_assert( constant-expression, string-literal );

The constant-expression parameter of a static_assert declaration represents a software assertion. A software assertion specifies a BOOLEAN condition that you expect to be true at a particular point in your program. 

        * If the boolean condition is TRUE   the static_assert declaration has no effect.
       *  If the boolean condition is false, the assertion fails, the compiler displays the message in string-literal parameter, and the compilation fails with an error. 

----------------------------------------------------------------------------------------------------------------
I  need to check  more the code,  but  I believed that in our Portapack case,  always the  
instantiation of 'typename T:  is referring to an array size  power of 2, not ?  
Therefore , when we are compiling this  line . static_assert ,  we alwasy will get TRUE, and  no impacct at all about our code generation. 

Then from that sense , it  seems  also OK to delele that particular  “safety compile debugging line” ?  

Then basically I can agree with Olewales  suggestion, lines 116, 117 : 
 
/*  static_assert(power_of_two(s.size()), "Array size must be power of 2");  */
   const size_t mask = s.size() - 1;  
 
I need to continue checking and studying about it , and confirming  that there is also no side effects on GCC9 and GCC10 with those two changes.

If anyone has some other alternative  better syntax solutions (regarding that static_assert declaration debugging test at compile time ) ,  it would  be highly  appreciated ! (compatible for both versions GCC9  / GCC10)

ArjanOnwezen, 2021-11-24:
Solved in new release with #419

closed 2021-11-24 by ArjanOnwezen