portapack-mayhem/mayhem-firmware #206: USB, LSB not working. Only DSB

catalinalb, 2020-10-03:
After updateing to v 1.2.3 from v 1.2.2, in Recieve->Audio->AM  I can select USB or LSB but the waterfall is still showing the DSB. 

Same with the CW mode.

This feature was working well on previos version. 

For those who don´t know USB is  Upper Side Band and LSB is Lower Side Band.  CW - verrry narrow band for Morse Code.

Thank you

eried, 2020-10-04:
Please make a video of the problem in 1.2.3 and show how it works on 1.2.2? there was no changes on that part in that version

catalinalb, 2020-10-04:
Yes I will do that but Is it safe to downgrade?

eried, 2020-10-04:
Yes, it is safe.

catalinalb, 2020-10-04:
Hi,
So I did install the version 1.2.2 and is the same. Good thing is that I can hear on the jack output that the selection is happening but it´s just about the waterfall that is not beeing shown properly. This was working before the update.
I have attached some screenshots that I´ve edited so you can understand better what I mean. Also here is the video showing the both versions: 
https://www.youtube.com/watch?v=xMDTA-DSbMI
I hope someone can address this issue.

Thank you.

[image]
[image]
[image]

eried, 2020-10-04:
In releases you can browse back old versions, if you find the one that works, please post the screenshot that looks like the one you remember. There I can check what actually changed

euquiq, 2020-10-04:
Hi @catalinalb ! What you are experiencing is correct. It's because of this pr:

https://github.com/eried/portapack-mayhem/pull/150

Although you can hear the correct mode, you now "see" the whole range at all times.

I wouldn't know with "all authority" if this is correct, but it "seems" nicer than just cut off the rest of the spectrum, "only" because you chose USB / LSB or CW. I would say that it is better to LISTEN into your selection, but SEE the whole range. 

Of course, maybe it is a matter of opinions. Maybe it needs to be reverted into the older more "visually impaired" version. At least now you know it is not an "issue" but it is "by design".

strijar, 2020-10-07:
That's right - now you can see the entire "band", but only the filtered area is heard. Misleading green filter stripe indicator. Over time, I will redo it so that it points correctly in different modes (DSB/LSB/USB/CW)

joyel24, 2020-11-20:
It can be a good idea to have a rectangle in the waterfall showing the selection of the corresponding mode in the spirit of this image (from a websdr).

joyel24, 2020-11-20:
It can be a good idea to have a rectangle in the waterfall showing the selection of the corresponding mode in the spirit of this image (from a websdr).
[https://i.imgur.com/XTWy3LGl.png](https://i.imgur.com/XTWy3LGl.png)

ArjanOnwezen, 2021-12-07:
In 1.4.x you can see a green stripe indicating USB/LSB/DSB 
[image]
Closing this issue, since USB/LSB mode is working fine

closed 2021-12-07 by ArjanOnwezen