portapack-mayhem/mayhem-firmware #207: wide band sweeper/analyzer

Kunkles, 2020-10-04:
im looking to repurpose my HackRF/Portapack on set (movies/TV) to help everyone manage frequencies and eliminate interference with eachother. There doesn't seem to be an efficient way to do this with the portapack just with it attached to a computer in HackRf mode.

 what would be most handy is 2.4/5ghz with channel indicators and then some way to set custom ranges (there are a few devices that use specific ranges but have limited channels)

any info or help is appreciated, or maybe the firmware does this and im missing somthing?

eried, 2020-10-04:
Hi, there is no built in tool for this (I do not think it would be hard to make), but what you could do for now is maybe use SDR# and add a custom bandplan.xml with your channels

Kunkles, 2020-10-05:
> Hi, there is no built in tool for this (I do not think it would be hard to make), but what you could do for now is maybe use SDR# and add a custom bandplan.xml with your channels

Thats great, i can venmo you something if that helps ;-) having something a little custom for this application would make things go a lot smoothing in a pinch

eried, 2020-10-05:
You should specify what you want and then you can open a bounty -> bountysource.com/teams/portapack-mayhem using this current link to your issue (https://github.com/eried/portapack-mayhem/issues/207). Someone might jump on to make it

Kunkles, 2020-10-06:
ok ill check it out, haven't ever done that before. would it still be a part of the main firmware? i wouldnt be opposed to paying for it to happen and still having it available to everyone

eried, 2020-10-06:
It can become part. But, you would not be paying for it directly, you would be just part of a bounty that has no guarantee on time or even if anyone will want to make it (ofc, more money in the bounty more easy to get people).

Please try to define exactly what would "close" your bounty: "A new app that display _________ and ____________ that could look like: --- sketch --- , etc"

Kunkles, 2020-10-06:
yeah ok sounds good. Ill be more specific

Kunkles, 2020-10-06:
how does this look?

Kunkles, 2020-10-07:
> It can become part. But, you would not be paying for it directly, you would be just part of a bounty that has no guarantee on time or even if anyone will want to make it (ofc, more money in the [...]

you think thats enough bounty? idk what the going rate would be for work like that?

eried, 2020-10-07:
It is ok as small incentive in case someone already wants a challenge. The screen needs to be portrait tho. I mean for a very slow fps could be sideways, but it would add a lot of more work.

Kunkles, 2020-10-07:
> It is ok as small incentive in case someone already wants a challenge. The screen needs to be portrait tho. I mean for a very slow fps could be sideways, but it would add a lot of more work.

ok roughly swapped it around, idk if you are interested in taking it on for more we could chat over email rkunkleman ( at ) gmail ( dot ) com

eried, 2020-10-07:
Join discord on the link on the main readme and you can discuss there :)

euquiq, 2020-10-21:
I just proposed this app, which may be what you are looking for (or at least it is similar), @Kunkles !

https://github.com/eried/portapack-mayhem/pull/215

I may add a text field (there is space for that) that consults a LOOKINGGLASS.TXT when moving the marker, and according to the frequency whereabouts, it can just tell you what's in there. 

Like: if the marker is at 2450Mhz, the txt will advise: 

`2436-2458 WIFI  CHANNEL 9` Maybe this will be a nice compromise between what you ask, and what hackrf is actually capable of doing ?

Cheers!

euquiq, 2020-11-14:
The Looking Glass App has been out for about three weeks. It would be nice to hear some feedback.  From my part, I suspect that the portapack / hackrf is capable of doing "about that" and not much more. 

I understand that the mock screens on this issue depicts a far more sophisticated interface, but I am afraid -in particular for the first screen- that it is not feasible. Besides, there are already several free apps for Android able to show the WIFI channels / signal strength / width out there.

For the second screen, Looking Glass "kind of" does that: it allows you to define and inspect different ranges. I would add that "visually" you get a good idea on how wide is a wifi channel, since you can move the frequency cursor along the different signals.

I would love to know if this issue can be "closed" and see if the Bounty attached to it is lost / partially / totally assigned. Or on the other hand, the issue will remain open, and if so, I suppose it will need some update on what is to be expected, against the current implementation.

Cheers!

Kunkles, 2020-11-14:
Hey sorry i've been busy with work. I'll take another look at it and see
what it's looking like. I'm not too worried about the mock screens. you had
a lot of the functionality there the last time i took a look. I really
appreciate all the time you've put into it so far

I'm not sure how to finalize/assign the bounty. If you know anything about
that let me know.

euquiq, 2020-11-14:
Apparently, one of the first steps is to close this as an issue. Is that about right, @eried  ?

closed 2020-11-14 by Kunkles

eried, 2020-11-14:
Yes, when the author (@Kunkles) is satisfied that the issue is fulfilled, he could close and he will receive some message from bountysource on how to continue. The system just piggybacks to github, but it is not really connected.

gullradriel, 2023-05-10:
Done with level app and level view in looking glass.