portapack-mayhem/mayhem-firmware #208: Portapack not showing right frequencies

loewal, 2020-10-06:
Hi all,

I am using Hackrf one + Portapack v1.2.3 in combination with OpenWebRX.
Perhaps not a question for this forum:
Why is the frequency in the display different and when tuning not showing that change?

eried, 2020-10-06:
The display only shows the center frequency. Try selecting a frequency far away from the current, the display should change.

loewal, 2020-10-06:
Yes I noticed, but selecting a frequency at the edge of the screen doesn’t change anything.
Something Soapy-related perhaps?

loewal, 2020-10-06:
And I would like the change on screen in OpenWebRX being reflected in the Portapak's screen of course.

loewal, 2020-10-06:
Like SDRangel, Shinysdr etc.

eried, 2020-10-06:
> Yes I noticed, but selecting a frequency at the edge of the screen doesn’t change anything.

If you tune to 1 ghz and then 2 ghz, does it change?

eried, 2020-10-06:
I am not sure, it would help if you record a video where you tune and it does not get reflected on the portapack screen, just to see what is happening. Maybe someone else could add some insights too. 

I know in SDR# works ok, but it does not update for small changes in frequency.

loewal, 2020-10-06:
Recording something that doesn't change is not very useful, isn't it?
I think Soapy is messing up, because Shinysdr also doesn't, but connecting it to SDRangel in macos goes perfect.

eried, 2020-10-06:
"Portapack not showing right frequencies" -> Then is not the portapack showing right frequencies? Just some software not working?

rascafr, 2020-10-06:
@loewal You can try `hackrf_transfer` with the `-r` option, the portapack *must* display the frequency you set in the command line.

OpenWebRX, Gqrx and other SDR software "hardware" receiver frequency is different than the one displayed at the cursor position; they perform software frequency translation before demodulation. 

(**Source: OpenWebRX author's thesis, page 30** https://sdr.hu/static/bsc-thesis.pdf)

Either this is the main reason, or `rtl_sdr` / soapy driver is not working properly

loewal, 2020-10-07:
Thnx François,
The frequency I enter with hackrf_transfer is indeed shown on the lcd.
Also connected the hackrf to my iMac: ok!
And in a Parallels virtual machine: ok!

I keep investigating....

rascafr, 2020-10-07:
Are you using OpenWebRX on your iMac? If you're using the docker image, you might not be able to make it work with the HackRF, since the `--device /dev/bus/usb` sharing parameter is not working "as it is" under OS X

Dirk275, 2020-10-25:
I don't know, possibly I had sth similar:
HackRF connected to my Laptop, running SDR# -> The HackRF Portapack sometimes didn't show the same frequency on the blue HackRF screen as tuned in SDR#. Also changes of gain ... were not displayed properly.
I tested the same with my PC: no problem.
Solution: it was a problem of bad USB power supply: I added an active USB hub for my laptop, then everything works fine.
But I'm not sure, that your problem is covered by my findings.

closed 2020-10-28 by eried