portapack-mayhem/mayhem-firmware #211: Frequency highlighting bug

Dirk275, 2020-10-09:
I found a small GUI bug:
In Receiver - Audio mod the frequency (MHz) is displayed for example in WFM like this:
97.5000
If I place the cursor on this frequency, then all numbers are highlighted.
If I move the cursor away from the frequency value, then the last number (0) stays highlighted, all other nunbers (97.500) are normal.

euquiq, 2020-10-09:
I can check and fix this

euquiq, 2020-10-09:
I can't reproduce the bug. I get into Receiver -> audio, change freq into -say- 90.3000, and move to the right or left, and I cannot get the last number to stay highlighted. Which firmware version are you using, Dirk275 ? Or what may I be doing wrong in order to reproduce the bug ? Maybe you are trying this in a firmware version that has been modified / customized ?

Dirk275, 2020-10-10:
Oh, I'm using v1.2.0 unmodified. I'll update to v1.2.3 soon and test again.

Dirk275, 2020-10-10:
Sorry,  now tested with v1.2.3 and everything is ok with frequency highlighting.

closed 2020-10-10 by Dirk275