portapack-mayhem/mayhem-firmware #217: TypeError: can only concatenate str (not "bytes") to str when creating map with Python script

OdinsHat, 2020-10-24:
**Describe the bug**
When executing ```python3 generate_world_map.bin.py```

I receive the error 
```
Traceback (most recent call last):
File "generate_world_map.bin.py", line 48, in <module>
line += struct.pack('H', pixel_lcd)
TypeError: can only concatenate str (not "bytes") to str
``` 

**To Reproduce**
cd to firmware/tools in the repo. The location of all the python scripts. Execute ```generate_world_map.bin.py``` 

**Expected behavior**
A worldmap.bin fiule (more specifically sdcard/ADSB/world_map.bin) should be generated. Instead the above bugf happens.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Current master
* Latest nightly
* It also happens with Python 2

eried, 2020-10-24:
Can you check this file: https://github.com/eried/portapack-mayhem/blob/47c5c1e9e53d38a92b8e8edffd4366d2a8c97a53/firmware/tools/generate_world_map.bin.py

Strangely the old one works fine in Windows WSL Ubuntu, but not on the native python3 x64

OdinsHat, 2020-10-25:
@eried this works perfectly. My apologies I do Python coding as a hobby - I should have caught this myself and made a pull request with the fix - it seems so obvious now. I suppose I've been PHP developing too long.

Anyway - the worldmap file is built but now it complains it can't be copied to a FAT32 formatted card. Is there another format the card can be done in? Be assured whatever your answer I will contribute it to the Wiki.

eried, 2020-10-25:
Why it cannot be copied?

OdinsHat, 2020-10-25:
The worldmap is 7.9Gb. FAT32 which is what you're meant to format the sdcard as can only manage files upto 4Gb. Unless there's another format is can be done in?

eried, 2020-10-26:
oh, you will need to resize the input map then. There is no other option for now.

closed 2020-10-28 by eried