portapack-mayhem/mayhem-firmware #219: 1.3.0 -  Speaker no longer working

miotislucifugis, 2020-11-02:
**Describe the bug**
Just upload 1.3.0 and I lost the sound from the internal speaker.   i can enable it with the icon, and  turn up the volume, but no sound.   using H1 portapack.    Im trying to find a copy of the previous release to compare and check that sound is indeed working.... and not a loose wire  ( i havent used the hackrf in a few weeks)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Audio'
2. Tap on 'speaker icon'  enable speaker.... turn up volume.
3. no audio from the speaker

**Expected behavior**
I expect the speaker to work

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release- 1.3.0
* 1.2.3 and previous version worked fine

eried, 2020-11-02:
If you go back to 1.2.3, you get the speaker back?

miotislucifugis, 2020-11-02:
Just tried it.  Speaker works fine on 1.2.3

eried, 2020-11-02:
Strange, I will set 1.3.0 on hold for testing

euquiq, 2020-11-02:
I tried 1.3.0 on my H1 and speaker works fine. Are you sure you are not on SPEC mode inside the audio app ?

JIC, also, did you try first to get into options->Interface and disable / re-enable the speaker ?

eried, 2020-11-02:
Could you post a video @miotislucifugis of 1.2.3 vs 1.3.0 doing the same?

miotislucifugis, 2020-11-02:
yeah, Im stupid.    Sorry for the false alarm.
  For the record, I didnt catch that receive->Audio was set on "Spec" (after coming from the looking glass app)  
If it is set to something audible, there is audio.   duh

closed 2020-11-02 by eried