portapack-mayhem/mayhem-firmware #221: POCSAG receiver seems to freeze

tomevans80, 2020-11-02:
**Describe the bug**
If POCSAG is opened while a transmission is occurring, it decodes it ok the first time, but then stops working.
Once transmission pauses, or if it's opened during a pause, it doesn't decode or log any later transmissions.

**Affected versions**
1.2.3

tomevans80, 2020-11-21:
I've also discovered that changing frequencies revives decoding, as long as you arrive onto the new frequency during an active transmisison.

chrisgxj, 2021-02-09:
I have found the exact same with POCSAG on mayhem firmware v1.3.0 how can I try to fix this myself?

ArjanOnwezen, 2021-11-09:
looks like issue has been solved by #426

tomevans80, 2021-11-18:
Thank you! Took me a while to remember how to compile it but I got there. 👍

ArjanOnwezen, 2021-11-24:
Issue solved in current release with #426

closed 2021-11-24 by ArjanOnwezen