portapack-mayhem/mayhem-firmware #223: Scanner frequentie file (scanner.txt) can't load up to more than 100 frequenties #173

Bracchiosaurus, 2020-11-03:
The titlel is saying all. 
Perhaps a little minor software adjustment possible? to solve this problem must be not a great problem due?
May be solved in the next firmware update?

tnx Ralph

GullCode, 2020-11-03:
Problem is well known now, it has to do with hardware limitations of the hackRF.
Usually you can load up to 500 freqs, if you cannot maybe your file is not well formatted or the SD card is not working.
Can you post your SCANNER.TXT file somewhere so we can have a look ?

GullCode, 2020-11-03:
As for the mitigation, I saw a way to do it, but it would slow down the scanner at some point (some SD reading while scanning).
Internal freq list can only be 500 items wide at max, so we would have to read freqs by blocks of X lines (actually 500) and wait for the scanning thread to process the block before reading another one.

Bracchiosaurus, 2020-11-05:
Hell0 GullCode,

Very generous of you too look at it.
The mentioned 500 lines will not executed it stuck on 100 diffrent frequencies which I manually add in the scanner.txt file.
The 100 frequencies are processed very well by HackRF and portapack shows the scanning display and scan process at max speed and than loops back to the first frequency when number 100 was reached and so on and on. It stops when activity is on one of my given 100 frequencies, So 100%. The .txt file is very well formatted otherwise it wouldn't run at all. The SD card (formatted FAT32) is working very well 'cause I've got the maps too see (show-up) when tracking plane(s) and the rest of it (see directory ADSB.) Perhaps there are limitations which had to do with the hackRF hardware (piping?, databus archiecture?, microprocessor limits?) I don't know. To fill or change the content of the .txt file I am using the program "Teksteditor" of Apple (working with MACOSX ver.10.11.6) and then when filled up (with 100 freq.) I transfer it via USB when portapack is coupled in HackRF mode on my macbook-pro, putting it directly in the "FREQMAN" directory. Then scanning performed very well. When do more frequencies (> then 100 per example 101) the frequencies above the 100 will ignored but scanning on the given 100 is still performed. That's All I have to say about it. TNX fr response.

Ralph

closed 2020-11-05 by Bracchiosaurus