portapack-mayhem/mayhem-firmware #224: How

luozhiwenchn, 2020-11-04:


luozhiwenchn, 2020-11-04:
It seems that long press the button in the middle will make it keep TX, but the device will restart

eried, 2020-11-04:
Restart after how long?

luozhiwenchn, 2020-11-04:
For 2 seconds.
I am not sure if it is right to long press the button in the middle to make it keep TX.

GullCode, 2020-11-05:
Is it really restarting ? Because the specs says 2 seconds on the left button or the middle button on arrow pad is making the device start or stop, not restarting. On my setup it's restarting only if I'm pressing the reset button on top of the portapack.

gerryk, 2020-11-05:
Not seeing the behaviour being described. 
This is on a Portapack H1
I select TX Activation: PTT, hold down the middle button in the jogwheel and it stays in transmit for as long as I hold the button.

luozhiwenchn, 2020-11-05:
Sorry.
I may have a problem with my words.
I am using H2, I am just wondering how to make it keep TX. Which button should I press and how long?

gerryk, 2020-11-05:
I think the problem is that on H2, there is some power management circuitry that is entirely in hardware, so not controllable from the firmware. One of these functions seems to be a soft power button, which is bound to the encoder pushbutton. This is defined in the software as the 'action' button, so changing this would be problematic. Any change to the button assignments on the encoder/direction buttons and action button would break the UI.
The touscreen TX key does work though, but I concede is awkward to use.

GullCode, 2020-11-05:
I tested on my H2  and yes, pushing 2 seconds on dial button or center button is making the device to power on or off.
Only good solution is to press the touchscreen, or make the microphone app support a single push to turn on TX and a single push to turn it off.

luozhiwenchn, 2020-11-05:
So there is no way to keep it TX if I am using H2 without the touchscreen?
It is necessary to find another way to keep it TX for H2.
As GullCode said, making the microphone app support a single push to turn on TX and a single push to turn it off is a good solution.

gerryk, 2020-11-05:
> So there is no way to keep it TX if I am using H2 without the touchscreen?

Not right now, that I can think of.

> It is necessary to find another way to keep it TX for H2.

This is difficult, since the code has no way to tell whether it is H1 or H2. 

> As GullCode said, making the microphone app support a single push to turn on TX and a single push to turn it off is a good solution.

This seems like the only option, apart from modifying the board. If it was me, I would mod the board to either have a separate button, or re-use another button as the power button. I get that this is not a workable solutino for most.

luozhiwenchn, 2020-11-05:
I think H1 or H2 can be chosen by the users.
Like Options-Interface-Hide H1 speaker option.
But I think making the microphone app support a single push to turn on TX and a single push to turn it off is the best.

gerryk, 2020-11-05:
Agreed, certainly the simplest, which almost always equates with best.

luozhiwenchn, 2020-11-05:
Hope all of these can be changed soon

luozhiwenchn, 2020-11-05:
@eried 
Thanks to all of you.

GullCode, 2020-11-05:
Just for the sake of it I asked my H2 provider if he can provide a way to shortcut the poweroff function on one of the buttons.
I'll tell you if he answers.

gerryk, 2020-11-05:
@luozhiwenchn I suggest adding to the description exactly what is needed. Generally speaking developers work with well-defined requirements and acceptance criteria.

@GullCode  
> Just for the sake of it I asked my H2 provider if he can provide a way to shortcut the poweroff function on one of the buttons.
> I'll tell you if he answers.

If I could get a decent high quality photo of the front and rear I could probably work something out. There are no schematics publicly available for the H2 unfortunately.

GullCode, 2020-11-05:
I'll see if I can do it. From what I see on mine the circuits are visibles but under a black paint that's covering the whole pcb

gerryk, 2020-11-05:
> 
> 
> I'll see if I can do it (if I find some time). From what I see on mine the circuits are visibles but under a black paint that's covering the whole pcb

Not too concerned about the soldermask... I'm curious to know what the traces from the centre button and the encoder button to on their way to whatever pin header they connect to.  Also interested in the detail around the power management IC.

gerryk, 2020-11-06:
XSX on Discord posted that the little man icon on the top bar activates TX toggle, so I tested it and it does. Funny how avan after having this thing for nearly 2 years I still learn things about it regularly.

closed 2020-11-06 by eried

GullCode, 2020-11-06:
Well. It is indeed activating kind of a toggle, but while blanking out screen and controls.
I think it's just due to the way the little guy is working, and it has nothing to do with the app itself.