portapack-mayhem/mayhem-firmware #225: The ability to decode dmr and p25 phase 1 and phase 2

craphole, 2020-11-07:
the ability to decode dmr and p25 phase 1 and phase 2 so that you don't have to carry around a laptop and use dsd (digital speech decoder)

craphole, 2020-11-12:
so is it possible for you to add that to the mayhem firmware?

jeremiah-k, 2020-11-13:
I use this function in SDRTrunk and love it. Is this the only issue opened on this? I'm considering putting a bounty on it.

craphole, 2020-11-14:
i can already decode dmr and p25 phase 1 and phase 2 with DSDFastlane the pro paid updated version of dsd digital speech decoder on my laptop but i want to be able to decode it on the go with the portapack mayhem firmware

msoose, 2020-11-23:
This would be a good addition, and doesn't necessarily need to support any vocoder for voice.

I'd toss in to a bounty for it for sure.

gschwiter, 2020-11-25:
This would be great to have the P25 phase 2 added. I have it built into my scanner in truck and home station however it would be great to take it on the go. it would be nice to be able to download the scan list from Radio Reference.

DomTheDorito, 2020-11-25:
And with the huge bandwidth the HackRF has, wouldn't it technically be possible to be able to do trunked systems?

hotaru2k3, 2021-05-12:
> And with the huge bandwidth the HackRF has, wouldn't it technically be possible to be able to do trunked systems?

I occasionally use sdrtrunk on my desktop with a HackRF to monitor a trunked P25 system, so it definitely should be possible.

ArjanOnwezen, 2022-04-21:
Duplicate of #360 .Other ticket has more info

closed 2022-04-21 by ArjanOnwezen