portapack-mayhem/mayhem-firmware #229: Tesla Charge Door Opener

eshattow, 2020-11-11:
Suggested because it's very simple, although it may be unlikely you'd just be carrying around a PortaPack H1 in your pocket and want to use it for any legitimate purpose to open your Tesla Charge Door.

https://github.com/fredilarsen/TeslaChargeDoorOpener

eried, 2020-11-11:
You can record the signal at 433 mhz and check if works. That .c16 file could be added to the SDFILES included with then releases

HackingIntoYourHeart, 2022-03-14:
https://github.com/HackingIntoYourHeart/Tesla-Charging-Port-Opener

Done.

eried, 2022-03-14:
Cool :) I will add them

jLynx, 2022-03-23:
Out of curiosity, why was this change reverted?

eried, 2022-03-24:
some people say it works, some say it does not. I think it will confuse users. For me personally, did not work

jLynx, 2022-03-24:
> some people say it works, some say it does not. I think it will confuse users. For me personally, did not work

We have been chatting about this today. I took my own recording that worked on my car and my friends. But it did not work on another car. But we think that's because the car was in deep sleep, and needs the phone near it to wake it up. That's my theory at the moment anyway.

ArjanOnwezen, 2022-03-24:
I can confirm @jLynx file is working in The Netherlands. That one was for 433 MHz. Perhaps you can do a PR for it.

jimilinuxguy, 2022-03-25:
[Archive.zip](https://github.com/eried/portapack-mayhem/files/8346668/Archive.zip)
Attached is a zip file containing two HackRF c16 captures and their corresponding frequency text files. These captures hve been tested (and captured) by me and are known by several others to be working.

ArjanOnwezen, 2022-03-25:
> [Archive.zip](https://github.com/eried/portapack-mayhem/files/8346668/Archive.zip) Attached is a zip file containing two HackRF c16 captures and their corresponding frequency text files. These captures hve been tested (and captured) by me and are known by several others to be working.

Hi @jimilinuxguy why are there two files with the same frequency? Do they both work on the same car?

ArjanOnwezen, 2022-03-25:
Previous files were pulled back

jimilinuxguy, 2022-03-25:
> Hi @jimilinuxguy why are there two files with the same frequency? Do they both work on the same car?

Its two captures that work. One from a Tesla charging station and another from a ChargePoint system.

jLynx, 2022-03-25:
[Tesla_Charge_Port_OpenNZ_AUS.zip](https://github.com/eried/portapack-mayhem/files/8348435/Tesla_Charge_Port_OpenNZ_AUS.zip)
Just to keep things in one place instead of being spread across Discord and GitHub, here is my capture that we have been talking about above

ArjanOnwezen, 2022-03-25:
> Its two captures that work. One from a Tesla charging station and another from a ChargePoint system.

So I guess one of them can then later be picked.
I will also try them later today.
The 433MHz one did already work fine.

jLynx, 2022-03-30:
Added a PR here https://github.com/eried/portapack-mayhem/pull/526

ArjanOnwezen, 2022-03-31:
True the files published work fine on 433MHz. Could someone in US check if those also work fine on 315MHz ?

ArjanOnwezen, 2022-04-01:
files have been made available https://github.com/eried/portapack-mayhem/tree/next/sdcard/SAMPLES :)

closed 2022-04-01 by ArjanOnwezen