portapack-mayhem/mayhem-firmware #230: Replaced missing parts on the H1 board yourself

kilohertz79, 2020-11-15:
I thought it could be a bug in my PCB of the H1-Portapack,because i have completed it with any parts that was not equipped such as the IC U6 for charging the LiIon pack and its external wiring.

I noticed that the **R7** must not be equipped at all, because then the display remains dark! ????
( connection between VBUS and VBUSCONTROL)

Does any of you know what tasks **R2** (#PDN) **C13** and ultimately the bridge **R7** have? Or is it better to remove these parts again?

Thanks in advance!

Dirk275, 2020-11-22:
R7: No clue
R2: holds PDN of AK4951 low (power down). If you equippe R2 without pulling PDN# high, the AK4951 will be down permanently.
C13: for SVDD, don't equippe.

pa0o, 2020-11-23:
hi does it function okay with accu after you placed U6 ?

kilohertz79, 2020-11-25:
> R7: No clue
> R2: holds PDN of AK4951 low (power down). If you equippe R2 without pulling PDN# high, the AK4951 will be down permanently.
> C13: for SVDD, don't equippe.

Thanks for your reply! I removed C13 , now I have to check the audio in "RECORD"-Mode......there was a tickling tone on the Loudspeaker since i placed U6.
And the USB-Powering made noise after installing U6....not good
I think it's 2 audioamplifires running in parallel that mades the noises......

kilohertz79, 2020-11-25:
> […](#)

Yes, the charging function is the only positive  since I placed U6
The audio problem troubles me really...... I think the H2-Boards from Banggood or alibaba are the better ones

Dirk275, 2020-11-27:
I think that U6 with it's functionality was not equipped on the H1, because it causes switching noise making the reception much worse.
On the H2 this is the same, but the noise is lower than on the H1 (better PCB layout?).

kilohertz79, 2020-11-28:
> I think that U6 with it's functionality was not equipped on the H1, because it causes switching noise making the reception much worse.
> On the H2 this is the same, but the noise is lower than on the H1 (better PCB layout?).

That can be, whatever....i will try to power down the U6 without the complete remove of it, or make it switchable....any idea?

Dirk275, 2020-11-28:
Perhaps the best option would be to charge the battery before using the HackRF. If it is powered by the battery alone, there should be less/no noise from U6.
Did you also equippe the GPS section?
I would be interested in this function, because of the known position: frequency tables and other region specific data could be used. You could also use AIS with your own position.....
But: this would mean a support from the firmware.

pa0o, 2020-11-30:
thanks for sharing interesting  issue about noise and H2 / H1

ArjanOnwezen, 2021-05-18:
@eried  Could you close this. Since it looks like a hardware mod issue