portapack-mayhem/mayhem-firmware #233: Spectrum painting

DomTheDorito, 2020-11-17:
Essentially something like this, but native on the portapack.

https://www.rtl-sdr.com/painting-on-the-rf-spectrum-with-a-hackrf/

xchgre, 2020-11-25:
Hey, the user  Fohdeesha at Reedit:
5 years ago
"For anyone curious this is really easy to do in windows with a computer-connected rig. Download Coagula, load an image in, and it'll spit out a wav file. simply play that wav back out through your radio, and you've transmitted an image that will show up on anyones waterfall."

Go ahead, create a folder with several wav and paint!

DomTheDorito, 2020-11-25:
Sure, but the idea was to be able to just insert an image and the portapack generates the .wav itself.

bernd-herzog, 2023-05-20:
Closed as this is now available

closed 2023-05-20 by bernd-herzog