portapack-mayhem/mayhem-firmware #234: Bluetooth Broken/Useless Record button

RFburn, 2020-11-19:
**Describe the bug**
Portapack locks up if you click the REC button on the BLE sniffer tool

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the BLE Tool under Receivers
2. Tap on Hit Record
3. Enjoy your free time until resetting the device

**Expected behavior**
That the record button wouldn't exist or just write MAC info to a TXT file somewhere (why I don't know, if we had GPS there may be some use to logging BLE and location, but just MAC addresses I don't see much reason other than to not lock up)

**Affected versions**
* Since introduction from what I've seen. I hit it, it locks up, just always been that way.

**Additional**
Highlights a reason to know what GPS module to solder in place and have its functionality integrated. Many forms of frequency detection at a particular location could be of great use. Maybe your city has a device that emits a particular frequency and are located around the city, well if you had the scanner, with a threshold, and option Log Detection Location to file, you could drive around detecting the location of said imaginary devices and plot them on google earth or the likes. Maybe a few other factors recorded with it, like SNR, signal strength, and the obvious time and date (which could also be set via GPS). Just a thought on something that would add tremendous amounts of additional ways the device could be used. That or if it supports USB Host, use an external GPS via USB or some UART pins so the GPS module could be either temporarily connected or a 3d printed shell could be used to attach the GPS antenna to the pack, maybe 45-90 degrees off the back so also acts like a mini kickstand, but this is unrelated to the error other than if made to work, a way the button on the BLE page could've of use.

I did search reports first, so hope this isn't a repeat, but Bluetooth returned no results of this error. Appologies ifsomehow

joyel24, 2020-11-20:
@RFburn 
I published a new version on my repo to fix this bug.
There is already a pull request in the eried repo here : #232 
You can also test it directly by downloading the firmware here [v1.3.0b](https://github.com/joyel24/portapack-mayhem/releases/tag/v1.3.0b)

eried, 2020-11-21:
Cool! For any next PR, please keep the old fix in its own PR and just create a new one if the new stuff has no relation (i.e. no relation with docker)

jamesshao8, 2020-12-08:
BLE is still experimental.
Most result can't pass crc check. Hence MAC address is likely to be mistaken by 1 or 2 bit.

euquiq, 2021-01-20:
Are there any improvements on BLE ? the BT ID (mac) data that this app is discovering on my Portapack has no resemblance with the real numbers I get on my Android BT scanner running on my phone.

As it is, more than WIP, it actually seems pretty much useless.

jamesshao8, 2021-01-21:
@euquiq I don't really have an android phone, hence not sure if your "Android BT scanner" is producing right result in the first place. My guess is your scanner on android also includes EDR device which isn't supported by portapack. I compared the result with the btle scan on ubuntu using "hcitool lescan". The result is in the wiki.

I don't think there will be future improvements from me on BLE. I made the code free and open source in hope that users would improve it by themselves. I don't have much responsibility on that. Feel free to improve or even remove it in future release. Same goes for analog tv rx, nrf24l01 rx, gps sim, etc.

gullradriel, 2023-03-31:
closed in favor of #791 which is having a wider description of the problem. 
I'll cross reference this.

commit 842e89e9 (2020-11-23): Merge pull request #232 from joyel24/next

Dockerfile Improvements + Fix #234 Removed record button in BTLE RX + update contributors
closed 2023-03-31 by gullradriel