portapack-mayhem/mayhem-firmware #235: No data from SD if starting app too quickly after portapack startup

GullCode, 2020-11-21:
**Describe the bug**
When you start up the portapack, if you click too quickly in an app that is loading datas from the SD card, the app is not having it's data.

**To Reproduce**
1. Start with a turned of portapack
2. Quickly click on an app that is loading datas from SD card to work (like the search app in my case). If you want to test you should try to quickly click on tools, freqman. It will say 'EMPTY SD CARD'. 
3. The datas are not present. Most of the time the app will start with default settings instead of recalling previous settings
4. As during that time the SD is finally mounted, just going back to main menu and into the given app is enough for the app to find the SD

**Expected behavior**
The portapack should wait for the SD to be ready before letting us start apps

**Affected versions**
Next, latest and previous version.

**Additional**
If you can not see the problem, try to shut off splash screen option. If you are keeping the startup splash screen it is hard to spot as it's leaving enough time for the SD to be mounted

GullCode, 2020-11-21:
That can be fixed, because the SD icon is turning from a yellow '?' to a green 'OK'. When it's green, apps are all working fine.

eried, 2020-11-21:
Interesting. Bad quality microsd cards might be impacting this too

eried, 2020-11-23:
I cannot replicate this, can you post a video? I tried as fast as possible to open Replay -> Open, and it always work.

GullCode, 2020-11-23:
Sure: https://youtu.be/OlEi7SLBAiU
Replay may not access files until you click on load. Freqman / Search app are directly looking into when starting

eried, 2020-11-23:
Does not happen with freqman, or filemanager :/ https://www.youtube.com/watch?v=wHWuCl5KXmI

GullCode, 2020-11-23:
My SD card is for sure slower than yours then. Or maybe because it's a big one it's taking more time to load ?

eried, 2020-11-23:
yes, maybe happens with slow sd cards... I have a 32 gb "year of the dog" banggood one (my favorite ones, bad experiences with crappy sandisk/kingston clones with are impossible to tell apart from clones). It would be nice if someone could replicate this

GullCode, 2020-11-23:
Mine is failing in Debug/SD card => test. First test says "fail: compare" and then stops.
Everything else is working (record/replay, file reading/saving in apps)

GullCode, 2020-11-23:
BTW it's a no name 32GB (so not that big in fact). I may try with a branded one to see the difference.

GullCode, 2020-12-14:
Well, for those who'll eventually have the same problem and find the closed issue:
Just wait until SD card icon is turning green before starting any app. That way you'll be sure the app will be able to read from the SD.

closed 2020-12-14 by GullCode

asergunov, 2023-11-29:
I have the same Fail: Compare but my card icon is perfectly green