portapack-mayhem/mayhem-firmware #238: Portapack H2 and almost no reception with the screen turned on

DiscoMan18, 2020-11-25:
Hi, here is video showing the problem: [https://youtu.be/bm6LQm8r18c](url)

Some time ago a opened an issue with the detailed description and steps to reproduce it. Here it is: #https://github.com/eried/portapack-mayhem/issues/9#issue-610215478

Since then I upgraded the firmware to the latest 1.3.0 but the problem remained the same. Could you take a look at it? 

Thanks in advance,
Alex.

eried, 2020-11-25:
What happens if you are in hackrf mode using SDR# or similar as tuner... does the screen also affects reception there?

DiscoMan18, 2020-11-25:
It's impossible to turn off the screen of Portapack while in HackRF mode. But yes, with the screen attached and turned on the reception is much worse even in HackRF mode (I used Cubic SDR to test it). However, without a portapack connected to the main board, the reception gets significantly better.

GullCode, 2020-11-25:
Turning off the main battery is helping as it's the charging that is causing flickers/bad reception (on my H2)

DiscoMan18, 2020-11-25:
Yes, the power/charging circuit can be a source of interference, but to my opinion the source of problems with the screen could be in the code, because it's a lot more noticeable when receiving FM stations in 88-108 mhz and requires a lot more processing power due to higher bandwidth. It's just my guess though.

eried, 2020-11-25:
Since is the portapack a good test would be to get another one from another seller to test. It might be a noisy charger controller, with the screen on consumption increments. There is no really code doing fine tunning with the screen and some specific freq range.

Robi2003, 2020-12-02:
Hey DiscoMan18,

I see you have the PortaPack H2, too.

I got it today. But I have a problem. It will not start. I plugged it into USB on my PC and into my battery pack.
Should I see something on the PortaPack H2 at least just an LED lighting up?

Because I think it is not working. Even my PC is not recognizing anything.

Sorry for my off-topic post.

Dirk275, 2020-12-03:
> Hey DiscoMan18,
>
> I see you have the PortaPack H2, too.
>
> I got it today. But I have a problem. It will not start. I plugged it into USB on my PC and into my battery pack.
> Should I see something on [...]

Without a first firmware update you will not see anything on your Portapack, if you bought it separately from the HackRF One.

Robi2003, 2020-12-03:
But for a firmware update the HackRF must be detected by the PC?

There is nothing. No recognizing / detection from Windows, no LED light when I plug it into a USB port.

GullCode, 2020-12-03:
Try the hackRF without the portapack on it. May be a faulty device. You should at least see a led indeed.

ArjanOnwezen, 2021-05-21:
@DiscoMan18  
In HackRF mode you can switch off the screen.
Downloaded the latest version (2021.05.01)from: https://downloads.myriadrf.org/builds/PothosSDR/
Using Mayhem 1.4.0
hackrf_debug.exe -u 0
will turn screen off
hackrf_debug.exe -u 1
will screen back on again

Please check reception..

closed 2021-05-21 by ArjanOnwezen