portapack-mayhem/mayhem-firmware #240: no voltage when outputing a freq

markl17, 2020-11-29:
**Describe the bug**
A clear and concise description of what the bug is. 
when generating a freq using gnuradio 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' python3 test.py
[test.py.zip](https://github.com/eried/portapack-mayhem/files/5612705/test.py.zip)

2. Tap on '....'
3. I used to get .012v of ac now I get nothing I think I burned out a chip or two

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Previous one (if any) that did not presented the issue
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

eried, 2020-11-30:
I imagine the output is the same using an older version, i.e. `OLD_2018_sharebrained_portapack-h1-firmware.bin` right?

markl17, 2020-11-30:
I shorted out some chip I don’t know which ones

markl17, 2020-11-30:
I definitely know I shorted something out but what I have no clue from outside device is acting perfectly except the voltmeter not showing anything

ArjanOnwezen, 2021-06-06:
@markl17  closing the issue, since it's a hardware  issue not effecting other units.

closed 2021-06-06 by ArjanOnwezen