portapack-mayhem/mayhem-firmware #243: CTCSS Decode on WFM

mwareman, 2020-11-30:
**Is your feature request related to a problem? Please describe.**
NFM has CTCSS Decode. Please port this over to the WFM module so that it is available there.

**Describe the solution you'd like**
CTCSS Decode should be on all analog audio modules. The code was written already and only implemented on NFM - so it should be relatively easy to port over.

**Describe alternatives you've considered**
Switching to NFM to view the CTCSS decode - but when doing this the received audio is no longer useful.

Brumi-2021, 2023-07-03:
Hi @mwareman , sorry for late reply . 
Technically is feasible to activate the CTCSS tone decoding to the WFM receiver , but as far as I know,  WFM is mainly used for commercial FM Broadcasting , and I do not know any spec usage of (WFM + CTCSS). 

So far all CTCSS  spec usages that I know are for PMR and analog Ham radio repeater all of them working I  NBFM . 
So in that sense, we do not see any real application usage  of that requested feature , but if you have other understanding pls let us know about it .

Cheers

eried, 2023-12-04:
I am closing issues without much discussion on going to keep this clean. Please feel free to reopen.

closed 2023-12-04 by eried