portapack-mayhem/mayhem-firmware #245: Custom Width Waterfall (compared to Rx Audio & Looking Glass)

ianrenton, 2020-12-01:
**Is your feature request related to a problem? Please describe.**
I would like to visually see received power over a range of a few MHz (e.g. 144-146 or 430-440). I don't think this is currently possible unless I've missed something? In the Rx Audio waterfall you can see a few kHz either side of your centre frequency, or in Looking Glass you can see a wide band with 240 MHz as the minimum width.

**Describe the solution you'd like**
 Ideally I'd like something in the middle, so I can look at a band and see a width of ~1-10 MHz displayed. The existing waterfall functionality would be fine, doesn't have to be a full-on frequency analyser as in #179 / #207.

**Describe alternatives you've considered**
Using the audio waterfall and scanning around with the thumb wheel; using Looking Glass and a magnifying glass!

**Additional**
This functionality seems to be exactly what the Looking Glass app was developed for, I'm wondering if there's a technical/hardware limitation that drives the 240 MHz minimum width - if not, maybe this minimum could just be reduced?

ianrenton, 2020-12-01:
Just went looking through the source code myself and noticed the comment on commit 210ec9dd1b - guess this will be a long term "no rush" kind of enhancement request!

euquiq, 2021-01-21:
Maybe I am missing something evident ... but what you are asking can be done inside the RX -> AUDIO app ... you can select the SPECTRUM view, and choose the bandwidth up to 20Mhz. It seems to be exactly what you need.

ianrenton, 2021-01-21:
No, you are absolutely right! I didn't spot that mode before, but it's exactly what I was after. Thanks

closed 2021-01-21 by ianrenton