portapack-mayhem/mayhem-firmware #246: intelligent RF jamming

thesle3p, 2020-12-07:
**Is your feature request related to a problem? Please describe.**
Modern RF systems have Jamming detection that triggers after RF jamming is detected for a predetermined period of time.

**Describe the solution you'd like**
Would it be possible to to implement a Variant of the Jamming application that runs on the specified frequency for a user defined period of time in seconds and then stop for a user defined cooling off period before running jamming again.

eried, 2020-12-08:
What kind of parameter would you expect from an improved jammer like this? i.e. which range of cool down and such?

thesle3p, 2020-12-08:
I would let the user specify the cool down period from one to 60 seconds.

euquiq, 2020-12-08:
@thesle3p in short, you need a "cooldown" selector, from 0 to 60, in seconds, added to the Jammer App, introducing a TX pause in the loop

thesle3p, 2020-12-09:
yes

lucalucazzo, 2020-12-10:
ciao. qualcuno puo aiutarmi a trovare istruzioni sul mio rf1 h2 con mayhem 1.3.0

lucalucazzo, 2020-12-12:
[image]
Hi I'm new.  I took hackrf H2 with mayhem 1.3.0 i would like to test the copy but i don't know how to do with this remote of my car.  Someone help me thanks

euquiq, 2020-12-12:
@lucalucazzo Hi. What you ultimately want, is not possible. I would recommend you to search and learn about rolling codes, and build your knowledge from those concepts. Which will explain you why you cannot just "copy" that remote's signal.

Anyway, Github is for issues in the CODE inside Mayhem firmware, so  I would suggest facebook, or the Discord chat server if you need help.

lucalucazzo, 2020-12-12:
The best software mayhem or avoc?  Maybe my hackrf one h2 is not good because it mounts mayhem?  Would he need avoc?

lucalucazzo, 2020-12-14:
hello friend thanks for the answer. now i wonder why he can copy this and send it back working?

[image]

lucalucazzo, 2020-12-14:
with this to work why?

GullCode, 2020-12-14:
Because that key fob is not using rolling codes
.

lucalucazzo, 2020-12-14:
So portapack doesn't copy Rolling code?

GullCode, 2020-12-14:
It can. But it will eventually only work once and only in case you recorded it while being out of the car range.

lucalucazzo, 2020-12-14:
How can I find the right frequency of the remote control?  Does Portapack have this function?  Thanks

BrunoKlu, 2020-12-14:
@lucalucazzo  Yes It does using the "looking glass" app.
You can tune the start and stop frequency of this swipe spectrum analyser (in your case try 200mhz to 1000mhz) Then just use very low LNA/VGA preamp settings to have a completely black\blue waterfall. Then hit the rf remote key button close to the portapack antenna and you will see a green>red dot on the waterfall. That is your frequency. Now move the marker on this dot to display the frequency.

lucalucazzo, 2020-12-14:
Can you show me a video?

BrunoKlu, 2020-12-14:
I have no vidéo at the moment, I can make one later today, now I'm going outside for a run. 🏃‍♂️

lucalucazzo, 2020-12-14:
[image]
What do I open?

BrunoKlu, 2020-12-14:
https://youtu.be/vFp82NBSbxs

Here is a quick Video

My car key was really low battery so I did put a bit of gain on hack rf. 

I found that there is a preset on the sd card called 315/433 Mhz keyfobs that you can use. 

Otherwise just set the start and stop frequency manually.

lucalucazzo, 2020-12-14:
In mine he only writes subaru on keyfob then asks for remote control codes over the frequency I send you photos.  But can I listen to my cell phone?  Is Mayhem better than avoc?
[image]

lucalucazzo, 2020-12-14:
How is the application called x understand the correct code

lucalucazzo, 2020-12-14:
[image]
What are these symbols?

BrunoKlu, 2020-12-14:
@lucalucazzo That is a lot of questions for this topic my friend. Did you manage to locate your key frequency ? It's usually around 433 in eu and 315 in us. Sometimes higher or lower in the spectrum.
The keyfob app is beta, and you will not do anything with it.
Try the capture and replay functions if you want to start recording analyzing and replay baseband. The portapack is not a toy that will unlock all the door ! You should Try learning about RF spectrum, rf signals analysis , modulation, digital coding first because that a very wide subject to learn here in this comments.

Cheers my friend

lucalucazzo, 2020-12-14:
Maybe mayhem not right?What are the symbols in the picture?  Man?

eried, 2020-12-14:
@lucalucazzo please ask your questions on facebook or discord

lucalucazzo, 2020-12-14:
I do not have Facebook

GullCode, 2020-12-14:
The read the wiki. Most of your questions are already answered there.

lucalucazzo, 2020-12-14:
BrunoKlu You can put the youtube link back it tells me removed

lucalucazzo, 2020-12-24:
Hello. why doesn't it turn on anymore?

https://user-images.githubusercontent.com/75666422/103068776-32019780-45be-11eb-88ad-78c4de52e139.mp4

eried, 2020-12-24:
Please @lucalucazzo, for chatting: ask in discord or facebook. If you have another issue create a new one and fill in details there. This issue is only for "intelligent RF jamming".

jamesshao8, 2021-01-06:
Hi, the original jamming program is able to automatically change center frequency, which isn't a constant signal.  Isn't it somehow intelligent? Why it's not good enough for modern RF system? @thesle3p

GullCode, 2021-01-06:
Maybe it's not adapted to frequency shifting jamming ?

thesle3p, 2021-01-06:
as I mentioned some systems will trigger an alarm if they detect RF interference for too long. to defeat that there would need to be a way to have the jammer run for a user defined period of time, stop jamming for another user defined period of time and then resume.

euquiq, 2021-01-06:
@thesle3p I am adding two "timer selectors": One will indicate the active jamming time (X seconds), and the other the cool-down time (Y seconds). So the jammer will output for N seconds, then cool down for Y seconds, and repeat. 

I think this is what you ask, is it OK ?

euquiq, 2021-01-07:
> **Is your feature request related to a problem? Please describe.**
> Modern RF systems have Jamming detection that triggers after RF jamming is detected for a predetermined period of time.
> 
> **Describe the solution you'd like**
> Would it be possible to to implement a Variant of the Jamming application that runs on the specified frequency for a user defined period of time in seconds and then stop for a user defined cooling off period before running jamming again.
> 
> There is a **[$500 open bounty](https://www.bountysource.com/issues/95023635-intelligent-rf-jamming?utm_campaign=plugin&utm_content=tracker%2F182166508&utm_medium=issues&utm_source=github)** on this issue. Add to the bounty at [Bountysource](https://www.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F182166508&utm_medium=issues&utm_source=github).

@thesle3p This new Jammer version incorporates two user-adjustable timers allowing for  jamming a defined period of time in seconds and then stop for a defined cooling off period before running jamming again.  

https://github.com/eried/portapack-mayhem/pull/263

thesle3p, 2021-01-07:
Great! I can't wait to test it

thesle3p, 2021-01-07:
confirmed it's working just as intended

closed 2021-01-07 by thesle3p

eried, 2021-01-08:
Awesome, can you report back any feedback @thesle3p? It will get merged into a release soon :)

thesle3p, 2021-01-08:
works like a dream