portapack-mayhem/mayhem-firmware #247: Touchscreen not activated when starting with USB plugged in the computer (H2)

GullCode, 2020-12-08:
**Describe the bug**
When plugged in the computer and starting up, touchscreen is not working on H2.

**To Reproduce**
Steps to reproduce the behavior:
1. Take a H2 portapack, verify it's off
2. Plug in USB from computer
3. Start up
4. Touch is not working at all

**Expected behavior**
Touch should work

**Affected versions**
Always had the bug whatever the version

**Additional**
If started from off and without USB wired on the computer, then touchscreen is working. Then if I'm plugging the USB computer wire, touch is still working.

eried, 2020-12-09:
It might be a noisy AC in your house, I cannot replicate this. Can you try in a laptop without being connected to the grid?

GullCode, 2020-12-09:
After some more tests:
-it's working if I'm directly plugged into the computer, but only if the cable is not too long (even shielded).
-What's still fun is that whatever I'm using to power it, if I'm starting before powering it with a wire, then touchscreen is working.

If it's okay for you we can close that issue. Maybe note it somewhere that if using a long cable or any usb wire extension the touchscreen can act funky or not work at all. Starting the portapack before pouring power from a long line source is always working.

eried, 2020-12-09:
how do you know it is shielded? what happens with that long cable if you use a laptop while on batteries?

GullCode, 2020-12-09:
Quick note for shielded: I bought them for that they were claiming to be shielded.
Now the battery is fully charger and I can not reproduce ! I think I found the culprit. I remember clearly seeing the green led indicating it was charging when I noticed the problem and made the issue.
I'm going to let it search for a while and drain the battery a bit, so I'll be sure that it's coming from charging. I'll confirm you the results.

eried, 2020-12-09:
HAHAHA we can make an entry in the wiki for "Gullcode". In case you are him, you may experience: ...

GullCode, 2020-12-09:
X-D
I would not be mad no worries ;-p

GullCode, 2020-12-10:
I can not reproduce the issue anymore. I don't know what was the problem. Closing !

closed 2020-12-10 by GullCode

janicektrnecka, 2021-01-07:
My battery charging circuit causes noisy issues too, especially notable in spectrum when you plug/unplug it from power. When the battery is full, or not charging the noise level drops significantly.