portapack-mayhem/mayhem-firmware #248: Romas

romkam8, 2020-12-10:
hello. where can i get portapack h2 mayhem instruction. i can't find a description of keyfob anywhere

eried, 2020-12-10:
Check https://github.com/eried/portapack-mayhem/wiki/

closed 2020-12-10 by eried

romkam8, 2020-12-10:
sorry but what i need is not there

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

eried, 2020-12-10:
There is no keyfob functionality. For remote controls you can use Record and Replay apps.

romkam8, 2020-12-10:
signal record, but when replayed, the car does not respond. Something goes
wrong with portapack or with me

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

euquiq, 2020-12-10:
What protocol is your test keyfob using ?